esphome/bluetooth-proxies #4: Question - esp32 proxy

adrianmihalko, 2022-09-05:
Hi,

Does this Bluetooth proxy let us communicate with BLE devices without actually having a Bluetooth dongle in my main HA system? 

Scenario:

- currently I am using one ESP32 with esphome (ble_client) as Bluetooth gateway for my HA system. Can this Bluetooth proxy  extend the range of this gateway?

GJurriens, 2022-09-05:
Yes, that is exactly what it is suppost to do. This wil make for a "distributed" bluetooth coverage system.
For more info check the video from DigiBlurDIY where @balloob joined if for addidtional info: https://www.youtube.com/watch?v=9WoWiWkOyS0

adrianmihalko, 2022-09-05:
This will be awesome. I installed it on a new ESP32 device, with the default config:

```
[22:01:22][C][wifi:358]:   Hostname: 'esp32-bluetooth-proxy-091d48'
[22:01:22][C][wifi:360]:   Signal strength: -65 dB ▂▄▆█
[22:01:22][C][wifi:364]:   Channel: 11
[22:01:22][C][wifi:365]:   Subnet: 255.255.255.0
[22:01:22][C][wifi:366]:   Gateway: 192.168.1.1
[22:01:22][C][wifi:367]:   DNS1: 192.168.1.10
[22:01:22][C][wifi:368]:   DNS2: 0.0.0.0
[22:01:22][C][logger:275]: Logger:
[22:01:22][C][logger:276]:   Level: DEBUG
[22:01:22][C][logger:277]:   Log Baud Rate: 115200
[22:01:22][C][logger:278]:   Hardware UART: UART0
[22:01:22][C][bluetooth_proxy:053]: Bluetooth Proxy:
[22:01:22][C][esp32_ble_tracker:717]: BLE Tracker:
[22:01:22][C][esp32_ble_tracker:718]:   Scan Duration: 300 s
[22:01:22][C][esp32_ble_tracker:719]:   Scan Interval: 320.0 ms
[22:01:22][C][esp32_ble_tracker:720]:   Scan Window: 30.0 ms
[22:01:22][C][esp32_ble_tracker:721]:   Scan Type: ACTIVE
[22:01:22][C][captive_portal:088]: Captive Portal:
[22:01:22][C][mdns:094]: mDNS:
[22:01:22][C][mdns:095]:   Hostname: esp32-bluetooth-proxy-091d48
[22:01:22][C][ota:085]: Over-The-Air Updates:
[22:01:22][C][ota:086]:   Address: esp32-bluetooth-proxy-091d48.local:3232
[22:01:22][C][api:138]: API Server:
[22:01:22][C][api:139]:   Address: esp32-bluetooth-proxy-091d48.local:6053
[22:01:22][C][api:143]:   Using noise encryption: NO
[22:01:22][C][improv_serial:032]: Improv Serial:
[22:06:05][D][esp32_ble_tracker:222]: Starting scan...
[22:11:05][D][esp32_ble_tracker:222]: Starting scan...
```

...but it doesn't seem to do anything else. Do I need to enable (and adopt in Esphome as a new bluetooth proxy) the following new config on my existing "main" ESP32 too? Main ESP32 is which is currently receiving data from Bluetooth clients. 

```
external_components:
  - source: github://pr#3736
    components: [api, bluetooth_proxy]
esp32_ble_tracker:
  scan_parameters:
    active: true
bluetooth_proxy:
```

nagyrobi, 2022-09-05:
You have to wait until HA 2022.9 will be released, properly, with docs.

GJurriens, 2022-09-05:
You should not have to wait for it. This is testing ready, so no reason not to test. Best not to try-out on your main. Reporting in this stage can actually help the devs find some last bugs if they exist.

balloob, 2022-09-06:
You have to wait for 2022.9 because it includes the ESPHome as Bluetooth proxy support. 

Closing issue because yes, that’s what it does. Just forwarding discovery packets for now. Active connections are in progress

closed 2022-09-06 by balloob