esphome/bluetooth-proxies #14: New ESP32 suggestion: AZ-Delivery DevkitC

Nardol, 2022-09-10:
Hello,

Could it be possible to add the `board: az-delivery-devkit-v4` to the compatible devices?

I've tested and it works as a bluetooth proxy.
I prefer not to create a PR because I won't be able to provide an image I will be sure of the Quality.
I used `esp32-generic.yaml` and only modified the board.

Nardol, 2022-09-11:
In the case it could be useful, I only have a French link but to be sure it is [this model](https://www.amazon.fr/gp/product/B071P98VTG/?tag=homeassista0e-20).

balloob, 2022-09-11:
There is no need to explicitly suggest this board, users can just use the generic ESP32 installer. We only add boards that we want to promote (ie M5Stack has a nice case) or that have Ethernet and are popular.

closed 2022-09-11 by balloob

Nardol, 2022-09-11:
Strangely I was not able to complete the configuration using improv, each time I typed my wifi settings and I had the message specifying it was not possible to connect to the wifi even if it connected so I thought it was because of the wrong board.
So I manually flashed it but it looks like using generic ESP32 does not work using the website button.