esphome/bluetooth-proxies #18: WT32-eth01

Eponas07, 2022-09-14:
have ordered the WT32-eth01 and wanted to flash this pber the website unfortunately I can no longer select this is no longer supported ? Thank you

balloob, 2022-09-14:
See #16. The YAML is still at https://github.com/esphome/bluetooth-proxies/blob/main/wt32-eth01.yaml and it will be added back in the future

closed 2022-09-14 by balloob

nagyrobi, 2022-09-14:
WT32-eth01 works perfectly as a proxy, it's just that you have to flash it manually atm, not via the website.

Eponas07, 2022-09-14:
Thanks for your answers Am unfortunately just starting, how can I flash the manually sorry Von meinem iPhone gesendet

nagyrobi, 2022-09-14:
Follow this thread: https://community.home-assistant.io/t/how-i-installed-esphome-on-the-wt32-eth01/359027

balloob, 2022-09-18:
The installer is now also available again on the website https://esphome.github.io/bluetooth-proxies/?diy

Eponas07, 2022-10-11:
Thx 👍Von meinem iPhone gesendet