esphome/bluetooth-proxies #19: lan eth doesn't work

645340633, 2022-09-17:
Hello guys，the GL-S10 I bought last week firmware is v2.1。I flash the esphome firmware but then the lan eth connect to the switch or router, there is no effect,   the router eth light doesn't flicker. please help what should I do.
[image]
[image]

645340633, 2022-09-17:
it looks like this

nagyrobi, 2022-09-19:
@blakadder

blakadder, 2022-09-19:
That's a v2.1 board apparently designed in may this year which cannot be same as mine since I got mine in February 2022. nothing now with Chinese manufacturers, different innards, model number still the same
 [image]

pvizeli, 2022-09-21:
My was an v2.3

blakadder, 2022-09-21:
geez, fast development cycle there, can you take a nice image so i can document it

PR #22 (2022-09-21): gl-s10 rev2.3 support
Update the config to support the latest GL S10 hardware revision.

Fixes #19
Fixes #21
closed 2022-09-21 by balloob