esphome/bluetooth-proxies #21: the  new version GL-S10 Ethernet chip  is IP101GRI that is not supported by esphome currently

645340633, 2022-09-19:
[image]
the new Ethernet chip of the board is IP101GRI that is not supported by esphome currently . I need Bluetooth proxy component through LAN eth. please      help me.

nagyrobi, 2022-09-19:
_New_ version? 
Please submit a feature request https://github.com/esphome/feature-requests/issues

nagyrobi, 2022-09-19:
@blakadder yours has a different Ethernet chip?

blakadder, 2022-09-19:
yes otherwise the configuration I made would not work

nagyrobi, 2022-09-19:
@645340633 shouldn't take too long at least for IP101: https://github.com/esphome/esphome/pull/3751

balloob, 2022-09-19:
I suggest we update the GL-S10 config to the newer model, as that is what people starting today will be using. We can add a link to the older YAML for people with an old model.

@blakadder is it as easy as just changing this line to IP101GRI? <https://github.com/esphome/bluetooth-proxies/blob/main/gl-s10.yaml#L23>

nagyrobi, 2022-09-19:
Likely not, as IP101 also needs a |power_pin`...
Gpios might be different...

blakadder, 2022-09-19:
Can't vouch the pins are same, I wouldn't change anything without a real world test first

balloob, 2022-09-19:
I know @pvizeli just got one so will check with him for an updated config.

pvizeli, 2022-09-20:
```
[10:33:19]entry 0x40080774
[10:33:19]I (62) boot: Chip Revision: 1
[10:33:19]I (63) boot_comm: chip revision: 1, min. bootloader chip revision: 0
[10:33:19]I (39) boot: ESP-IDF v3.3.2-323-gbf0220609-dirty 2nd stage bootloader
[10:33:19]I (39) boot: compile time 00:41:19
[10:33:19]I (40) boot: Enabling RNG early entropy source...
[10:33:19]I (45) boot: SPI Speed      : 80MHz
[10:33:19]I (50) boot: SPI Mode       : DIO
[10:33:19]I (54) boot: SPI Flash Size : 4MB
[10:33:19]I (58) boot: Partition Table:
[10:33:19]I (61) boot: ## Label            Usage          Type ST Offset   Length
[10:33:19]I (68) boot:  0 otadata          OTA data         01 00 0000b000 00002000
[10:33:19]I (76) boot:  1 phy_init         RF data          01 01 0000d000 00001000
[10:33:19]I (83) boot:  2 cert             WiFi data        01 02 0000e000 00008000
[10:33:19]I (91) boot:  3 nvs              WiFi data        01 02 00016000 0002a000
[10:33:19]I (98) boot:  4 ota_0            OTA app          00 10 00040000 001e0000
[10:33:19]I (106) boot:  5 ota_1            OTA app          00 11 00220000 001e0000
[10:33:19]I (113) boot: End of partition table
[10:33:19]I (118) boot_comm: chip revision: 1, min. application chip revision: 0
[10:33:19]I (125) esp_image: segment 0: paddr=0x00220020 vaddr=0x3f400020 size=0x34b30 (215856) map
[10:33:19]I (197) esp_image: segment 1: paddr=0x00254b58 vaddr=0x3ffbdb60 size=0x049dc ( 18908) load
[10:33:19]I (203) esp_image: segment 2: paddr=0x0025953c vaddr=0x40080000 size=0x06adc ( 27356) load
[10:33:19]I (213) esp_image: segment 3: paddr=0x00260020 vaddr=0x400d0020 size=0xf5330 (1004336) map
[10:33:19]I (506) esp_image: segment 4: paddr=0x00355358 vaddr=0x40086adc size=0x14574 ( 83316) load
[10:33:19]I (552) boot: Loaded app from partition at offset 0x220000
[10:33:19]I (552) boot: Disabling RNG early entropy source...
[10:33:19]I (553) psram: This chip is ESP32-D0WD
[10:33:19]I (557) spiram: Found 64MBit SPI RAM device
[10:33:19]I (562) spiram: SPI RAM mode: flash 80m sram 80m
[10:33:19]I (567) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
[10:33:19]I (574) cpu_start: Pro cpu up.
[10:33:19]I (578) cpu_start: Application information:
[10:33:19]I (583) cpu_start: Project name:     gl-s10-v3_0_3
[10:33:19]I (588) cpu_start: App version:      3.0.3
[10:33:19]I (593) cpu_start: Compile time:     Jul  7 2022 11:34:05
[10:33:19]I (599) cpu_start: ELF file SHA256:  d988f62e4dd85f27...
[10:33:19]I (605) cpu_start: ESP-IDF:          v4.2.3-98-g85337dc428
[10:33:19]I (611) cpu_start: Starting app cpu, entry point is 0x40081e28
[10:33:19]I (0) cpu_start: App cpu up.
[10:33:20]I (1118) spiram: SPI SRAM memory test OK
[10:33:20]I (1118) heap_init: Initializing. RAM available for dynamic allocation:
[10:33:20]I (1118) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[10:33:20]I (1125) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[10:33:20]I (1131) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[10:33:20]I (1137) heap_init: At 3FFCB4A8 len 00014B58 (82 KiB): DRAM
[10:33:20]I (1143) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[10:33:20]I (1150) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[10:33:20]I (1156) heap_init: At 4009B050 len 00004FB0 (19 KiB): IRAM
[10:33:20]I (1162) cpu_start: Pro cpu start user code
[10:33:20]I (1167) spiram: Adding pool of 4096K of external SPI memory to heap allocator
[10:33:20]I (1188) spi_flash: detected chip: generic
[10:33:20]I (1188) spi_flash: flash io: dio
[10:33:20]I (1189) cpu_start: Starting scheduler on PRO CPU.
[10:33:20]I (0) cpu_start: Starting scheduler on APP CPU.
[10:33:20]I (1197) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
[10:33:20]../components/gl_dev_nvs/src/gl_dev_nvs.c[39]: ***NVS INIT***
[10:33:20]../components/gl_dev_nvs/src/gl_dev_nvs.c[553]: This is the 12 boot
[10:33:20]I (1307) system_api: Base MAC address is not set
[10:33:20]I (1307) system_api: read default base MAC address from EFUSE
[10:33:20]../main/src/app_sync_info.c[49]: The flag of the data synchronization status was not found !!!
[10:33:20]I (1327) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[10:33:20]../components/gl_dev_protocol/src/gl_dev_protocol.c[42]: BLE MAC: 0CDC7E9AD832
[10:33:20]../components/gl_dev_protocol/src/gl_dev_protocol.c[46]: p_aes128cbc_config->key = goodlife--00BF0B
[10:33:20]I (1357) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[10:33:20]I (1367) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[10:33:20]I (1377) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[10:33:20]I (1397) wifi:wifi driver task: 3ffd7b1c, prio:23, stack:3584, core=0
[10:33:20]I (1407) wifi:wifi firmware version: e2a0df7
[10:33:20]I (1407) wifi:wifi certification version: v7.0
[... 15 lines cut ...]
[10:33:20]I (1467) wifi_init: tcp rx win: 5744
[10:33:20]I (1467) wifi_init: tcp mss: 1440
[10:33:20]I (1467) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
[10:33:20]I (1567) wifi:mode : sta (0c:dc:7e:9a:d8:30)
[10:33:20]../main/src/app_dev_info.c[177]: WIFI_EVENT_STA_START
[10:33:20]../components/gl_dev_network/src/gl_dev_network.c[66]: Get efuse device hardware version: 2
[10:33:20]I (1627) esp_eth.netif.glue: 0c:dc:7e:9a:d8:33
[10:33:20]I (1627) esp_eth.netif.glue: ethernet attached to netif
[... cut ...]
[10:33:25]../main/src/app_mqtt_info.c[248]: p_gl_mqtt_client_userConfig->password:
[10:33:25]../main/src/app_mqtt_info.c[60]: Other event id:7
[10:33:25]E (5897) TRANS_TCP: [sock=54] delayed connect error: Connection reset by peer
[10:33:25]E (5907) MQTT_CLIENT: Error transport connect
[10:33:25]../main/src/app_mqtt_info.c[57]: MQTT_EVENT_ERROR
[10:33:25]../main/src/app_mqtt_info.c[36]: MQTT_EVENT_DISCONNECTED
[10:33:25]ble cmd list create
[10:33:25]mqtt cmd list create
[10:33:25]use esp timer !!!
```

pvizeli, 2022-09-21:
```yaml
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO5
```

this work. It seems they used it from the dev reference board.

But not sure how we should deal with multiple hardware ref on this repository @balloob

blakadder, 2022-09-21:
v.2.1 as default is the sensible option, I'm updating my tutorial about the differences and set yaml config like this:

```yaml
## Configuration fo V2.1 hardware revision
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO5

## Comment the above and use this instead for V1.0 revision of the hardware
# ethernet:
#   type: LAN8720
#   mdc_pin: GPIO23
#   mdio_pin: GPIO18
#   clk_mode: GPIO17_OUT
#   phy_addr: 1
```

pvizeli, 2022-09-21:
> We don't deal with it. All new users have 2.1 so that's what we should offer. We can link to YAML in git history for old config

I got an 2.3

pvizeli, 2022-09-21:
I'm a bit unsure about `clk_mode: GPIO17_OUT` the reference board use `clk_mode: GPIO0_OUT` but some chips can use `GPIO16` or `GPIO17` as well. For me, it works with `GPIO17`, so it may use the ESP32 that can select this for clk. If we see issues at runtime, we should change it to `GPIO0`

balloob, 2022-09-21:
Opened a PR #22 which we can merge when clk has been decided 👍

pvizeli, 2022-09-21:
Okay, you can merge it @balloob it is GPIO17

closed 2022-09-21 by balloob

Bascht74, 2022-12-29:
@pvizeli I have ordered some new directly from GL.iNet and got V2.1. 
I have the same problem with ethernet as you. It is not stable, i cannot upload any new firmware to ist. It takes a long time to count some percent and then there is a timeout error...
Did you find any solution? what about the clk_mode gpio? Could that help?

oywino, 2023-02-01:
I have the GL-S10 v2.1 board. Several attempts to flash it using @blakadder 's page resulted in massive packet loss. My last attempt to flash with Blackadder's page (a couple of days ago) gave me v2022.12.6 allthough v2022.12.8 was already officially released.
So I tried instead building a few various bin files manually while experimenting with different YAML settings; IDF and Arudino, as well as changing the Ethernet speed settings, but no noticeable improvements. Still pretty useless.
So I went back to 2022.10.0 and set the Ethernet ports to 10MF and the packet losses went down from 25%+ to some 4-7%, low enough to be able to OTA. But doing so brings me back up to the latest and again packet losses are unacceptable.

[Question]: Is there a way to build bin file with my own YAML, using an ESPHome firmware version of my choice?

Bascht74, 2023-02-01:
@balloob Should be open another issue or could you reopen this issue as it is still not working OK?

HeFeng1947, 2023-03-31:
Hi guys, 

I am an employee of glinet and we are concerned about the feedback about the packet loss on eth issue. We have done a series of tests, and here are our findings.

-  After IDFv5.0, there is a problem with the IP101 driver that causes packet loss.
- In IDFv4.x, using esp32 internal clock to synchronize IP101 will cause packet loss.
- In IDFv4.x, using an external crystal to synchronize ESP32 and IP101 will not cause packet loss.
- Our gl-s10 is developed with stability in mind, even if it increases the cost, we provide an external crystal.

I haven't had time to confirm which clocking scheme ESPHOME is using, so if it's because of this problem, switching the configuration will fix it.
@oywino

nagyrobi, 2023-03-31:
@HeFeng1947 Thank you. There's a big confusion at the moment among our users also about the difference between the various hardware revisions of GL-S10. It would be very beneficial if you could help us determining clearly the configuration differences.

HeFeng1947, 2023-03-31:
> @HeFeng1947 Thank you. There's a big confusion at the moment among our users also about the difference between the various hardware revisions of GL-S10. It would be very beneficial if you could help us determining clearly the configuration differences.

First of all I'm sorry that we switched the phy selection last year and it caused some confusion. This is because the price of the original lan8720 has increased tens of times and the delivery time is not guaranteed at all. We have written a specific hardware version number in the EFUSE section and the software can distinguish between the two different hardware versions.
We will further investigate the esphome related compilation process in the next week and hope to give some help to the esphome community!

I am a new guy in ESPHOME community, is there a forum or something like a communication area in the community? I'd like to start a thread to communicate with people about the progress.

PR #22 (2022-09-21): gl-s10 rev2.3 support
Update the config to support the latest GL S10 hardware revision.

Fixes #19
Fixes #21