esphome/bluetooth-proxies #24: GL-S10 issues

pvizeli, 2022-09-22:
Okay. The network driver settings are 100% correct. But It crash. With enabled LED for BT it crash faster and without it run longer more stable. The original firmware using ESP-IDF but we can't because ESPHome have no ethernet support for it currently. But the network seems to have some issue as well, I lose sporadic packages (all 10 - 15 sec one package).

I tested a bit around, it seems that `board: esp32dev` works like the other definition on the board. I have no idea where I should look at. Maybe @blakadder have some ideas what also have changed and make issues?

pvizeli, 2022-09-22:
I tested now `board: nodemcu-32s` with LEDs. Currently, I had no crashes, package lose is the same. I think that is a driver issue or an issue with my device. But as long they do not crash, it's "fine"

I led it now run longer and see what happening

pvizeli, 2022-09-22:
Update, once I connect HA to it, it start crashing again:
```
10:13:09][10:13:09]
[10:13:09]rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
[10:13:09]configsip: 0, SPIWP:0xee
[10:13:09]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:13:09]mode:DIO, clock div:2
[10:13:09]load:0x3fff0018,len:4
[10:13:09]load:0x3fff001c,len:1044
[10:13:09]load:0x40078000,len:10124
[10:13:09]load:0x40080400,len:5828
[10:13:09]entry 0x400806a8
[10:13:10][I][logger:243]: Log initialized
[10:13:10][C][status_led:014]: Setting up Status LED...
[10:13:10][C][ota:465]: There have been 0 suspected unsuccessful boot attempts.
[10:13:10][D][esp32.preferences:113]: Saving 1 preferences to flash...
[10:13:10][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:13:10][I][app:029]: Running through setup()...
[10:13:10][D][binary_sensor:034]: 'Reset Button': Sending initial state OFF
[10:13:11][D][esp32_ble_tracker:264]: Starting scan...
[10:13:11][C][ethernet:039]: Setting up Ethernet...
[10:13:11][I][ethernet:092]: Starting ethernet connection
[10:13:22][I][ethernet:103]: Connected via Ethernet!
[10:13:22][C][ethernet:264]:   IP Address: 192.168.10.175
[10:13:22][C][ethernet:265]:   Hostname: 'gl-s10-bt-proxy-9ad830'
[10:13:22][C][ethernet:266]:   Subnet: 255.255.255.0
[10:13:22][C][ethernet:267]:   Gateway: 192.168.10.1
[10:13:22][C][ethernet:278]:   DNS1: 192.168.10.30
[10:13:22][C][ethernet:279]:   DNS2: 0.0.0.0
[10:13:22][C][ethernet:282]:   MAC Address: 0C:DC:7E:9A:D8:33
[10:13:22][C][ethernet:283]:   Is Full Duplex: YES
[10:13:22][C][ethernet:284]:   Link Up: YES
[10:13:22][C][ethernet:285]:   Link Speed: 100
[10:13:22][C][ota:089]: Over-The-Air Updates:
[10:13:22][C][ota:090]:   Address: gl-s10-bt-proxy-9ad830.local:3232
[10:13:22][C][api:025]: Setting up Home Assistant API server...
[10:13:22][I][app:062]: setup() finished successfully!
[10:13:22][I][app:102]: ESPHome version 2022.9.0 compiled on Sep 22 2022, 10:06:54
[10:13:22][I][app:104]: Project esphome.bluetooth-proxy version 1.0
[10:13:22][C][status_led:019]: Status LED:
[10:13:22][C][status_led:020]:   Pin: GPIO32
[10:13:22][C][logger:275]: Logger:
[10:13:22][C][logger:276]:   Level: DEBUG
[10:13:22][C][logger:277]:   Log Baud Rate: 115200
[10:13:22][C][logger:278]:   Hardware UART: UART0
[10:13:22][C][gpio.output:010]: GPIO Binary Output:
[10:13:22][C][gpio.output:011]:   Pin: GPIO14
[10:13:22][C][gpio.output:012]:   Inverted: YES
[10:13:22][C][gpio.output:010]: GPIO Binary Output:
[10:13:22][C][gpio.output:011]:   Pin: GPIO12
[10:13:22][C][gpio.output:012]:   Inverted: YES
[10:13:22][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Reset Button'
[10:13:22][C][gpio.binary_sensor:016]:   Pin: GPIO33
[10:13:22][C][bluetooth_proxy:053]: Bluetooth Proxy:
[10:13:23][C][safe_mode.button:022]: Safe Mode Button 'Safe Mode Boot'
[10:13:23][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[10:13:23][C][esp32_ble_tracker:776]: BLE Tracker:
[10:13:23][C][esp32_ble_tracker:777]:   Scan Duration: 300 s
[10:13:23][C][esp32_ble_tracker:778]:   Scan Interval: 1100.0 ms
[10:13:23][C][esp32_ble_tracker:779]:   Scan Window: 1100.0 ms
[10:13:23][C][esp32_ble_tracker:780]:   Scan Type: ACTIVE
[10:13:23][C][esp32_ble_tracker:781]:   Continuous Scanning: True
[10:13:23][C][ethernet:146]: Ethernet:
[10:13:23][C][ethernet:264]:   IP Address: 192.168.10.175
[10:13:23][C][ethernet:265]:   Hostname: 'gl-s10-bt-proxy-9ad830'
[10:13:23][C][ethernet:266]:   Subnet: 255.255.255.0
[10:13:23][C][ethernet:267]:   Gateway: 192.168.10.1
[10:13:23][C][ethernet:278]:   DNS1: 192.168.10.30
[10:13:23][C][ethernet:279]:   DNS2: 0.0.0.0
[10:13:23][C][ethernet:282]:   MAC Address: 0C:DC:7E:9A:D8:33
[10:13:23][C][ethernet:283]:   Is Full Duplex: YES
[10:13:23][C][ethernet:284]:   Link Up: YES
[10:13:23][C][ethernet:285]:   Link Speed: 100
[10:13:23][C][ethernet:148]:   Power Pin: GPIO5
[10:13:23][C][ethernet:149]:   MDC Pin: 23
[10:13:23][C][ethernet:150]:   MDIO Pin: 18
[10:13:23][C][ethernet:151]:   Type: IP101
[10:13:23][C][mdns:094]: mDNS:
[10:13:23][C][mdns:095]:   Hostname: gl-s10-bt-proxy-9ad830
[10:13:23][C][ota:089]: Over-The-Air Updates:
[10:13:23][C][ota:090]:   Address: gl-s10-bt-proxy-9ad830.local:3232
[10:13:23][C][api:138]: API Server:
[10:13:23][C][api:139]:   Address: gl-s10-bt-proxy-9ad830.local:6053
[10:13:23][C][api:143]:   Using noise encryption: NO
[10:18:10][I][ota:109]: Boot seems successful, resetting boot loop counter.
[10:18:10][D][esp32.preferences:113]: Saving 1 preferences to flash...
[10:18:10][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:18:11][D][esp32_ble_tracker:264]: Starting scan...
[10:21:04][D][api:102]: Accepted ::FFFF:C0A8:A1E
[10:21:04][D][api:102]: Accepted ::FFFF:C0A8:A1E
[10:21:04][W][api.connection:943]: aioesphomeapi (::FFFF:C0A8:A1E): Connection reset
[10:21:04][D][api.connection:861]: Home Assistant 2022.9.5 (::FFFF:C0A8:A1E): Connected successfully
[10:21:16]abort() was called at PC 0x401ced1b on core 1
[10:21:16]
[10:21:16]ELF file SHA256: 0000000000000000
[10:21:16]
[10:21:16]Backtrace: 0x4008f94c:0x3ffcc990 0x4008fbc9:0x3ffcc9b0 0x401ced1b:0x3ffcc9d0 0x401ced62:0x3ffcc9f0 0x401ce67d:0x3ffcca10 0x401ce754:0x3ffcca30 0x400d5855:0x3ffcca50 0x400d58b0:0x3ffcca70 0x400d5c31:0x3ffcca90 0x400d5cd9:0x3ffccb80 0x400d7a77:0x3ffccba0 0x401d79c1:0x3ffccca0 0x401d7a65:0x3ffcccc0 0x400da78a:0x3ffccce0 0x400dcc2e:0x3ffccd10 0x400e0720:0x3ffccd30 0x40090c3e:0x3ffccd50
[10:21:16]
[10:21:16]Rebooting...
[10:21:16]ets Jun  8 2016 00:22:57
[10:21:16]
```

Is it an issue with the network package lose that crash then the instance with an ongoing communication?

pvizeli, 2022-09-27:
It seems to be an issue with the old arduino framework that esphome use. I have the same issues also with other ESP-WROOM-32U/E models

Go0oSer, 2022-10-03:
I have same issue.

Lampyridae1984, 2022-10-05:
Same here, but only for one my units. Using "esp32doit-devkit-v1" for both.

phaeton82, 2022-10-08:
Same issue here.  It looks like the device reboots every minute or so when using the BIN from this website. Tried the works: different ethernet cable, POE switch, changed board types, disabled leds (works better indeed). Cannot do an OTA update so have to flash via the pins all the time. Anyone have more ideas?

tony-fav, 2022-10-08:
I'm unable to ota but I've got a semi stable build using the default interval and window settings for esp32_ble_tracker instead of 1100ms for both. 
[image]

nagyrobi, 2022-10-08:
I was about to suggest the same thing. How do you connect to the network? Ethernet, or WiFi?

tony-fav, 2022-10-08:
I'm on ethernet with the V2.3 pins. I've got "GL-S10-V201" with date stamp "2022/05/06"

tony-fav, 2022-10-09:
Get the occasional API disconnect and reconnect with error seeming to be related to TCP stack buffer filling.