esphome/bluetooth-proxies #27: Request for LilyGO T-Internet-COM support

unmesh59, 2022-09-30:
It looks very much like the T-Internet-POE board that drops POE but adds an embedded USB-to-serial converter that is wired to its USB-C port

https://github.com/Xinyuan-LilyGO/T-Internet-COM

Thanks

nagyrobi, 2022-09-30:
This should be moved to https://github.com/esphome/feature-requests

@unmesh59 the LAN chip is supported. If you have such a module, you can try [starting from the doc](https://esphome.io/components/ethernet.html?highlight=LILYGO) example configs to set your GPIOs and see if it works. I think you need to add `power_pin` and use another `mdc_pin`. I can't test as I don't have such a module.

unmesh59, 2022-09-30:
I did not realize it wasn't supported by ESPHome yet and will post the request there. 

I haven't done any flashing of ESPHome except for Bluetooth Proxy on a NodeMCU ESP32 which did not need me to understand much about what was happening!

Thanks.

closed 2022-09-30 by unmesh59