esphome/bluetooth-proxies #33: [active] is an invalid option for [bluetooth_proxy]. Please check the indentation.

ziolelle, 2022-10-05:
Using ESPHOME 2022.9.7

MY Config:
```
substitutions:
  name: esp32-bluetooth-proxy-xxxxxxxxxx
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: "xxxxxxxxxxxxxxxxxxx"
  password: "xxxxxxxxxxxxxxxxxxxxxx"
  fast_connect: on

# Enable logging
logger:
  level: DEBUG

switch:
  - platform: restart
    name: "${name} Restart"

sensor:
  - platform: wifi_signal
    name: "${name} WiFi Signal"
    update_interval: 60s
```

trying to validate the config in HomeAssistant AddOn or also using command line esphome, the config gives:

[image]

DarkosGahan, 2022-10-05:
Confirmed ! same here since yesterday

nagyrobi, 2022-10-05:
Wait till next release, please.

Grey-Lancaster, 2022-10-05:
+1
just getting in line here so I will get notified of the new update
Thanks,
Grey

balloob, 2022-10-06:
New version of ESPHome has been released that fixes this.

closed 2022-10-06 by balloob

marcelglaeser, 2022-10-06:
ESPHOME is updated to version 2022.9.3?

marcelglaeser, 2022-10-06:
> I am still getting the error?
> 
> substitutions:
> 
>   name: atom-bluetooth-proxy-d673a8
> 
> packages:
> 
>   esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main
> 
> esphome:
> 
>   name: ${name}
> 
>   name_add_mac_suffix: false
> 
> 
> 
> 
> 
> wifi:
> 
>   ssid: !secret wifi_ssid
> 
>   password: !secret wifi_password

ESPHOME is updated to version 2022.9.3?

Grey-Lancaster, 2022-10-06:
My bad.  I thought there was a new version of BTHome so I flashed it again.
Upgrade ESPHome and it works