esphome/bluetooth-proxies #36: Active Bluetooth connections not working with August Smart Lock Gen 3's

liquidbear99, 2022-10-06:
After installing the up to date Active Bluetooth version the lock does not use or connect to the functional bluetooth proxy.

pilonull, 2022-10-06:
Can confirm this is also happening to me with homekit integration on the Level Lock platform.  Lock works great when I'm using the on-board bluetooth adapter, but as soon as I enable active connections on esphome devices, the lock stops working. When I set them back to passive connections and only rely on the on-board BT adapter, the lock works great again.

Have this in my esphome config:

```
esp32_ble_tracker:
bluetooth_proxy:
  active: true
```

Home assistant has these errors in log: 

```
Logger: homeassistant.core
Source: components/esphome/bluetooth/client.py:66
First occurred: 11:01:01 AM (1 occurrences)
Last logged: 11:01:01 AM

Error executing service: <ServiceCall lock.unlock (c:01GEPWXHZWWD1GN9JD7DQ8CMVR): entity_id=['lock.kitchen_door_lock_lock_mechanism']>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 437, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 436, in send_message_await_response_complex
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 64, in _async_wrap_bluetooth_operation
    return await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 317, in read_gatt_char
    return await self._client.bluetooth_gatt_read(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/client.py", line 538, in bluetooth_gatt_read
    resp = await self._connection.send_message_await_response_complex(
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 439, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.BluetoothGATTReadRequest'>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1756, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 207, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/lock.py", line 114, in async_unlock
    await self._set_lock_state(STATE_UNLOCKED)
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/lock.py", line 118, in _set_lock_state
    await self.async_put_characteristics(
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/entity.py", line 91, in async_put_characteristics
    return await self._accessory.put_characteristics(payload)
  File "/usr/src/homeassistant/homeassistant/components/homekit_controller/connection.py", line 616, in put_characteristics
    results = await self.pairing.put_characteristics(characteristics)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/pairing.py", line 110, in _async_wrap
    return await func(self, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 76, in _async_wrap
    return await func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/pairing.py", line 789, in put_characteristics
    await self._async_request_under_lock(
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/pairing.py", line 250, in _async_request_under_lock
    pdu_status, result_data = await ble_request(
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 101, in ble_request
    return await _read_pdu(client, decryption_key, handle, tid)
  File "/usr/local/lib/python3.10/site-packages/aiohomekit/controller/ble/client.py", line 137, in _read_pdu
    data = await client.read_gatt_char(handle)
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 482, in read_gatt_char
    return await self._backend.read_gatt_char(char_specifier, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 52, in _async_wrap_bluetooth_connected_operation
    return await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/esphome/bluetooth/client.py", line 66, in _async_wrap_bluetooth_operation
    raise asyncio.TimeoutError(str(err)) from err
asyncio.exceptions.TimeoutError: Timeout waiting for response for <class 'api_pb2.BluetoothGATTReadRequest'>

```

```
Logger: aiohomekit.controller.ble.pairing
Source: runner.py:119
First occurred: 11:01:40 AM (1 occurrences)
Last logged: 11:01:40 AM

A2075GRD [MAC ADDRESS REDACTED] (id=MAC ADDRESS REDACTED): Failed to fetch disconnected events: Timeout waiting for response for <class 'api_pb2.BluetoothGATTReadRequest'>; rssi=-79
```

sarabveer, 2022-10-07:
The ESP32 is crashing, here is a log while connected via serial.

```
[I][esp32_ble_client:058]: Attempting BLE connection to <AUGUST_BT_MAC>
[I][esp32_ble_client:142]: Service UUID: 0x1800
[I][esp32_ble_client:143]:   start_handle: 0x1  end_handle: 0x5
[I][esp32_ble_client.service:057]:  characteristic 0x2A00, handle 0x3, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A01, handle 0x5, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 0x1801
[I][esp32_ble_client:143]:   start_handle: 0x6  end_handle: 0x6
[I][esp32_ble_client:142]: Service UUID: 0x180A
[I][esp32_ble_client:143]:   start_handle: 0x7  end_handle: 0xf
[I][esp32_ble_client.service:057]:  characteristic 0x2A29, handle 0x9, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A24, handle 0xb, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 0xFE24
[I][esp32_ble_client:143]:   start_handle: 0x10  end_handle: 0x20
[I][esp32_ble_client:142]: Service UUID: 0000003E-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x21  end_handle: 0x39
[I][esp32_ble_client:142]: Service UUID: 000000A2-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x3a  end_handle: 0x3f
[I][esp32_ble_client.service:057]:  characteristic 00000037-0000-1000-8000-0026BB765291, handle 0x3c, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x3f, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 00000055-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x40  end_handle: 0x4e
[I][esp32_ble_client:142]: Service UUID: 00000045-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x4f  end_handle: 0x5f
[I][esp32_ble_client:142]: Service UUID: 00000044-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x60  end_handle: 0x68
[I][esp32_ble_client.service:057]:  characteristic 00000019-0000-1000-8000-0026BB765291, handle 0x62, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000037-0000-1000-8000-0026BB765291, handle 0x65, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x68, properties 0x2
[D][esp32_ble_tracker:264]: Starting scan...
[I][esp32_ble_client:058]: Attempting BLE connection to <AUGUST_BT_MAC>
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[W][esp32_ble_client:089]: connect to 00:00:00:00:00:00 failed, status=133
[D][esp32_ble_tracker:264]: Starting scan...
[I][esp32_ble_client:058]: Attempting BLE connection to <AUGUST_BT_MAC>
[I][esp32_ble_client:142]: Service UUID: 0x1800
[I][esp32_ble_client:143]:   start_handle: 0x1  end_handle: 0x5
[I][esp32_ble_client.service:057]:  characteristic 0x2A00, handle 0x3, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A01, handle 0x5, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 0x1801
[I][esp32_ble_client:143]:   start_handle: 0x6  end_handle: 0x6
[I][esp32_ble_client:142]: Service UUID: 0x180A
[I][esp32_ble_client:143]:   start_handle: 0x7  end_handle: 0xf
[I][esp32_ble_client.service:057]:  characteristic 0x2A29, handle 0x9, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A24, handle 0xb, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A25, handle 0xd, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A26, handle 0xf, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 0xFE24
[I][esp32_ble_client:143]:   start_handle: 0x10  end_handle: 0x20
[I][esp32_ble_client.service:057]:  characteristic BD4AC611-0B45-11E3-8FFD-0800200C9A66, handle 0x12, properties 0xc
[I][esp32_ble_client.service:057]:  characteristic BD4AC612-0B45-11E3-8FFD-0800200C9A66, handle 0x14, properties 0x20
[I][esp32_ble_client.service:057]:  characteristic BD4AC613-0B45-11E3-8FFD-0800200C9A66, handle 0x17, properties 0xc
[I][esp32_ble_client.service:057]:  characteristic BD4AC614-0B45-11E3-8FFD-0800200C9A66, handle 0x19, properties 0x20
[I][esp32_ble_client.service:057]:  characteristic BD4AC615-0B45-11E3-8FFD-0800200C9A66, handle 0x1c, properties 0x14
[I][esp32_ble_client.service:057]:  characteristic BD4AC616-0B45-11E3-8FFD-0800200C9A66, handle 0x1f, properties 0x14
[I][esp32_ble_client:142]: Service UUID: 0000003E-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x21  end_handle: 0x39
[I][esp32_ble_client.service:057]:  characteristic 00000023-0000-1000-8000-0026BB765291, handle 0x23, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000014-0000-1000-8000-0026BB765291, handle 0x26, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000020-0000-1000-8000-0026BB765291, handle 0x29, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000021-0000-1000-8000-0026BB765291, handle 0x2c, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000030-0000-1000-8000-0026BB765291, handle 0x2f, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000052-0000-1000-8000-0026BB765291, handle 0x32, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 000000A6-0000-1000-8000-0026BB765291, handle 0x35, properties 0x2a
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x39, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 000000A2-0000-1000-8000-0026BB765291
[... cut ...]
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x68, properties 0x2
[D][esp32_ble_tracker:264]: Starting scan...
abort() was called at PC 0x401e257f on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f6e4:0x3ffd45c0 0x4008f961:0x3ffd45e0 0x401e257f:0x3ffd4600 0x401e25c6:0x3ffd4620 0x401e1ee1:0x3ffd4640 0x401e1fb8:0x3ffd4660 0x400dac41:0x3ffd4680 0x401389e1:0x3ffd46a0 0x40133b02:0x3ffd46c0 0x400909d6:0x3ffd46f0

Rebooting...
ets Jun  8 2016 00:22:57
```

sarabveer, 2022-10-07:
After reboot/crash, the same thing happens:

```
[I][esp32_ble_client:058]: Attempting BLE connection to <AUGUST_BT_MAC>
[D][esp32_ble_client:099]: [<AUGUST_BT_MAC>] Unexpected conn_id in CONNECT_EVT: param conn=0, open conn=32156
[I][esp32_ble_client:142]: Service UUID: 0x1800
[I][esp32_ble_client:143]:   start_handle: 0x1  end_handle: 0x5
[I][esp32_ble_client.service:057]:  characteristic 0x2A00, handle 0x3, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A01, handle 0x5, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 0x1801
[I][esp32_ble_client:143]:   start_handle: 0x6  end_handle: 0x6
[I][esp32_ble_client:142]: Service UUID: 0x180A
[I][esp32_ble_client:143]:   start_handle: 0x7  end_handle: 0xf
[I][esp32_ble_client.service:057]:  characteristic 0x2A29, handle 0x9, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A24, handle 0xb, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A25, handle 0xd, properties 0x2
[I][esp32_ble_client.service:057]:  characteristic 0x2A26, handle 0xf, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 0xFE24
[I][esp32_ble_client:143]:   start_handle: 0x10  end_handle: 0x20
[I][esp32_ble_client.service:057]:  characteristic BD4AC611-0B45-11E3-8FFD-0800200C9A66, handle 0x12, properties 0xc
[I][esp32_ble_client.service:057]:  characteristic BD4AC612-0B45-11E3-8FFD-0800200C9A66, handle 0x14, properties 0x20
[I][esp32_ble_client.service:057]:  characteristic BD4AC613-0B45-11E3-8FFD-0800200C9A66, handle 0x17, properties 0xc
[I][esp32_ble_client.service:057]:  characteristic BD4AC614-0B45-11E3-8FFD-0800200C9A66, handle 0x19, properties 0x20
[I][esp32_ble_client.service:057]:  characteristic BD4AC615-0B45-11E3-8FFD-0800200C9A66, handle 0x1c, properties 0x14
[I][esp32_ble_client.service:057]:  characteristic BD4AC616-0B45-11E3-8FFD-0800200C9A66, handle 0x1f, properties 0x14
[I][esp32_ble_client:142]: Service UUID: 0000003E-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x21  end_handle: 0x39
[I][esp32_ble_client.service:057]:  characteristic 00000023-0000-1000-8000-0026BB765291, handle 0x23, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000014-0000-1000-8000-0026BB765291, handle 0x26, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000020-0000-1000-8000-0026BB765291, handle 0x29, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000021-0000-1000-8000-0026BB765291, handle 0x2c, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000030-0000-1000-8000-0026BB765291, handle 0x2f, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000052-0000-1000-8000-0026BB765291, handle 0x32, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 000000A6-0000-1000-8000-0026BB765291, handle 0x35, properties 0x2a
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x39, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 000000A2-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x3a  end_handle: 0x3f
[I][esp32_ble_client.service:057]:  characteristic 00000037-0000-1000-8000-0026BB765291, handle 0x3c, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x3f, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 00000055-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x40  end_handle: 0x4e
[I][esp32_ble_client.service:057]:  characteristic 0000004C-0000-1000-8000-0026BB765291, handle 0x42, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 0000004E-0000-1000-8000-0026BB765291, handle 0x45, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 0000004F-0000-1000-8000-0026BB765291, handle 0x48, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000050-0000-1000-8000-0026BB765291, handle 0x4b, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x4e, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 00000045-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x4f  end_handle: 0x5f
[I][esp32_ble_client.service:057]:  characteristic 00000023-0000-1000-8000-0026BB765291, handle 0x51, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 0000001D-0000-1000-8000-0026BB765291, handle 0x54, properties 0x2a
[I][esp32_ble_client.service:057]:  characteristic 0000001E-0000-1000-8000-0026BB765291, handle 0x58, properties 0x2a
[I][esp32_ble_client.service:057]:  characteristic 000000A5-0000-1000-8000-0026BB765291, handle 0x5c, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x5f, properties 0x2
[I][esp32_ble_client:142]: Service UUID: 00000044-0000-1000-8000-0026BB765291
[I][esp32_ble_client:143]:   start_handle: 0x60  end_handle: 0x68
[I][esp32_ble_client.service:057]:  characteristic 00000019-0000-1000-8000-0026BB765291, handle 0x62, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic 00000037-0000-1000-8000-0026BB765291, handle 0x65, properties 0xa
[I][esp32_ble_client.service:057]:  characteristic E604E95D-A759-4817-87D3-AA005083A0D1, handle 0x68, properties 0x2
[D][esp32_ble_tracker:264]: Starting scan...
abort() was called at PC 0x401e257f on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f6e4:0x3ffd4560 0x4008f961:0x3ffd4580 0x401e257f:0x3ffd45a0 0x401e25c6:0x3ffd45c0 0x401e1ee1:0x3ffd45e0 0x401e1fb8:0x3ffd4600 0x400dac41:0x3ffd4620 0x401389e1:0x3ffd4640 0x40133b02:0x3ffd4660 0x400909d6:0x3ffd4690

Rebooting...
ets Jun  8 2016 00:22:57
```

amwalters, 2022-10-08:
Confirming the same issue with level lock. Error logs are exactly the same as posted above.

infocus13, 2022-10-09:
Can confirm same issue.

Home Assistant 2022.10 connected to an ESPHome device setup as BT proxy. August Gen 3 locks. Above behaviour is confirmed - locks do not connect and ESPHome continuously reboots.

sarabveer, 2022-10-10:
Could this PR be related? https://github.com/esphome/esphome/pull/3425

danlo315, 2022-10-13:
Want to follow up and confirm this issue.  HA docker running 2022.10.1 with ESPhome 2022.9.4 flashed via browser.  On both Assure Lock v2 and v1.