esphome/bluetooth-proxies #39: GL-S10 OTA issue

gerard33, 2022-10-08:
I have the GL-S10 with V2.1 board and initially flashed the firmware via ESPHome flasher.
That results in the GL-S10 device being discovered by Home Assistant, but I can't configure it in ESPHome. It's shown there as offline allthough I can open and read the logs in ESPHome.

I have done the initial install via ESPHome flasher because when using the official web installer the device doesn't get an IP address (DHCP on router is enabled). I solved this by adding a fixed IP address in the yaml file and upload that via ESPHome flasher.

When doing an OTA firmware update via ESPHome after that it returns the error below.

Besides that the devices becomes unavailable and then available again and that repeats every few minutes.

```
INFO Uploading /data/gl-s10-bt-proxy/.pioenvs/gl-s10-bt-proxy/firmware.bin (1493568 bytes)
Uploading: [== ] 3%
ERROR Error sending data: [Errno 104] Connection reset by peer
```
And when connecting ESPHome flasher this is shown in the logs. At the time the connection is lost, there is a watchdog error.
```
[21:02:08][C][mdns:095]: Hostname: gl-s10-bt-proxy-9ad9fc
[21:02:08][C][ota:089]: Over-The-Air Updates:
[21:02:08][C][ota:090]: Address: 192.168.1.101:3232
[21:02:20][D][ota:143]: Starting OTA Update from ::FFFF:C0A8:1A7...
[21:02:21][D][ota:312]: OTA in progress: 0.1%
[21:02:22][D][ota:312]: OTA in progress: 1.8%
[21:02:24][D][ota:312]: OTA in progress: 2.3%
[21:02:29]E (32270) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[21:02:29]E (32270) task_wdt: - loopTask (CPU 1)
[21:02:29]E (32270) task_wdt: Tasks currently running:
[21:02:29]E (32270) task_wdt: CPU 0: IDLE0
[21:02:29]E (32270) task_wdt: CPU 1: IDLE1
[21:02:29]E (32270) task_wdt: Aborting.
[21:02:29]abort() was called at PC 0x401b52c8 on core 0
[21:02:29]
[21:02:29]ELF file SHA256: 0000000000000000
[21:02:29]
[21:02:29]Backtrace: 0x400901dc:0x3ffbfe50 0x40090459:0x3ffbfe70 0x401b52c8:0x3ffbfe90 0x4008e679:0x3ffbfeb0 0x401e157f:0x3ffbc100 0x401b6ca3:0x3ffbc120 0x40092c8d:0x3ffbc140 0x400914ce:0x3ffbc160
[21:02:29]
[21:02:29]Rebooting...
[21:02:29]ets Jun 8 2016 00:22:57
[21:02:29]
[21:02:29]rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
[21:02:29]configsip: 0, SPIWP:0xee
[21:02:29]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:02:29]mode:DIO, clock div:2
[21:02:29]load:0x3fff0018,len:4
[21:02:29]load:0x3fff001c,len:1044
[21:02:29]load:0x40078000,len:8896
[21:02:29]load:0x40080400,len:5828
[21:02:29]entry 0x400806ac
```

balloob, 2022-10-09:
We have a bug where you can't OTA while bluetooth proxy is active. That's why we added a button to trigger safe mode. It will allow you to boot ESPHome without the bluetooth proxy active and the OTA will succeed.

closed 2022-10-09 by balloob

tony-fav, 2022-10-09:
I've had no luck with OTA on the GL-S10 even with using safe boot.

balloob, 2022-10-09:
If it still fails in safe mode, please open a new issue