esphome/bluetooth-proxies #41: Generic ESP32 proxy not working with Mi Flora

Lampyridae1984, 2022-10-09:
Home Assistant: 2022.10.1
Supervisor: 2022.09.1
Operating System: 9.0
Frontend: 20221006.0

Current ESP32 yaml config:

```
substitutions:
  name: "esp32-bluetooth-proxy"

esphome:
  name: ${name}
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
logger:

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5
  password: !secret ota_password_esp32_bluetooth_proxy

improv_serial:
captive_portal:

esp32_ble_tracker:
  scan_parameters:
    interval: 320ms
    window: 320ms
    continuous: true
    active: true

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic
```

My main and for the moment only goal (I have no other BT devices), was to connect my Mi Flora sensors through the proxy instead of a BT-adapter. Using my adapter (ZEXMTE BT-505) and the Xiaomi BLE integration works almost flawlessly, but I can't get them to update properly or being "discovered" by the proxy. As I mentioned, removing them doesn't seem to get them rediscovered (even after a system reboot). Not sure tho if the discovering issue has to do with the proxy or my Home Assistant setup.

**_Some_** of the individual sensors (light, temperature, moisture, conductivity and battery) update sometimes when I restart Home Assistant, but never after that. Below is a copy of my proxys log when (!) it kinda works after a system reboot. Has anyone had the same experience?

```
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy.yaml...
INFO Starting log output from esp32-bluetooth-proxy.local using esphome API
INFO Successfully connected to esp32-bluetooth-proxy.local
[13:50:57][I][app:102]: ESPHome version 2022.9.4 compiled on Oct  9 2022, 13:46:02
[13:50:57][C][wifi:502]: WiFi:
[13:50:57][C][wifi:360]:   Local MAC: 78:21:84:7C:A8:6C
[13:50:57][C][wifi:361]:   SSID: 'TB_MW_1303'[redacted]
[13:50:57][C][wifi:362]:   IP Address: 192.168.50.10
[13:50:57][C][wifi:364]:   BSSID: 24:4B:FE:30:62:E0[redacted]
[13:50:57][C][wifi:365]:   Hostname: 'esp32-bluetooth-proxy'
[13:50:57][C][wifi:367]:   Signal strength: -58 dB ▂▄▆█
[13:50:57][C][wifi:371]:   Channel: 8
[13:50:57][C][wifi:372]:   Subnet: 255.255.255.0
[... 254 lines cut ...]
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x0002, handle 0x15, properties 0x2
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x0004, handle 0x17, properties 0x12
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x0007, handle 0x19, properties 0x8
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x0010, handle 0x1b, properties 0x8
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x0013, handle 0x1d, properties 0xa
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x0014, handle 0x1f, properties 0x2
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x1001, handle 0x21, properties 0x10
[13:54:10][I][esp32_ble_client:142]: Service UUID: 0xFEF5
[13:54:10][I][esp32_ble_client:143]:   start_handle: 0x23  end_handle: 0x30
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 8082CAA8-41A6-4021-91C6-56F9B954CC34, handle 0x25, properties 0xa
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 724249F0-5EC3-4B5F-8804-42345AF08651, handle 0x27, properties 0xa
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 6C53DB25-47A1-45FE-A022-7C92FB334FD4, handle 0x29, properties 0x2
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 9D84B9A3-000C-49D8-9183-855B673FDA31, handle 0x2b, properties 0xa
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 457871E8-D516-4CA1-9116-57D0B17B9CB2, handle 0x2d, properties 0xe
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 5F78DF94-798C-46F5-990A-B3EB6A065C88, handle 0x2f, properties 0x12
[13:54:10][I][esp32_ble_client:142]: Service UUID: 0x1204
[13:54:10][I][esp32_ble_client:143]:   start_handle: 0x31  end_handle: 0x39
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x1A00, handle 0x33, properties 0xa
[13:54:10][I][esp32_ble_client:143]:   start_handle: 0x3a  end_handle: 0x42
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x1A11, handle 0x3c, properties 0x2
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x1A10, handle 0x3e, properties 0x1a
[13:54:10][I][esp32_ble_client.service:057]:  characteristic 0x1A12, handle 0x41, properties 0x2
[13:54:10][D][esp32_ble_tracker:264]: Starting scan...
[13:54:11][I][bluetooth_proxy:206]: [c4:7c:8d:6d:36:72] Disconnecting.
```

Lampyridae1984, 2022-10-09:
And just to be clear, the proxy is nowhere near any type of electronic interference, like my router of anything else that could cause issues with BT or WiFi.

nagyrobi, 2022-10-09:
Try decreasing `interval` to, say, 550ms.

Lampyridae1984, 2022-10-09:
Mmmm, not sure. Initially it seemed to have helped a bit, but after a while I basically got the same result. 

Some of them have all of their sensors exposed and some have only a few, or non at all. The most problematic seems to be the battery sensors, it's pretty much 50/50.