esphome/bluetooth-proxies #42: xiaomi_mjyd02yla not getting any status updates

ptrsconsulting, 2022-10-09:
I've been using esphome for quite a while now connecting with a few Mi thermometers and lately I've set up 2 mjyd02yl-a night ligths and I can't make it work. 

Here's my yaml:
`esphome:
  name: esp
  platform: ESP32
  board: esp-wrover-kit

sensor:
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:C6:XX:XX"
    # bindkey: "xxxxf0a2ec874d47af1e5fbc999c7bxx"
    temperature:
      name: "Lounge Temperature"
    humidity:
      name: "Lounge Humidity"
    battery_level:
      name: "Lounge Battery Level"
    signal_strength:
      name: "Lounge Signal"
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:10:XX:XX"
    # bindkey: "xxxx24c5bb2bc933abc9b0567177dexx"
    temperature:
      name: "Office Temperature"
    humidity:
      name: "Office Humidity"
    battery_level:
      name: "Office Battery Level"
    signal_strength:
      name: "Office Signal"
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:52:XX:XX"
    # bindkey: "xxxx92185b71778d818f22dc0e1eadxx"
    temperature:
      name: "Bedroom Temperature"
    humidity:
      name: "Bedroom Humidity"
    battery_level:
      name: "Bedroom Battery Level"
    signal_strength:
      name: "Bedroom Signal"
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:78:XX:XX"
    # bindkey: "xxxx2ca00d5bc901f9a845f8f78496xx"
    temperature:
      name: "Bathroom Temperature"
    humidity:
      name: "Bathroom Humidity"
    battery_level:
      name: "Bathroom Battery Level"
    signal_strength:
      name: "Bathroom Signal"

binary_sensor:
  - platform: xiaomi_mjyd02yla
    name: "Bedroom Night Light 1"
    mac_address: "68:AB:BC:4C:XX:XX"
    bindkey: "xxxx8de7c0c5dxx85727568c07238bxx"
    idle_time:
      name: "Bedroom Night Light 1 Idle Time"
    light:
      name: "Bedroom Night Light 1 Light Status"
    battery_level:
      name: "Bedroom Night Light 1 Battery Level"
    illuminance:
      name: "Bedroom Night Light 1 Illuminance"
  - platform: xiaomi_mjyd02yla
    name: "Bedroom Night Light 2"
    mac_address: "68:AB:BC:4D:XX:XX"
    bindkey: "xxxx348e6xxd58a0beceb1685afea2xx"
    idle_time:
      name: "Bedroom Night Light 2 Idle Time"
    light:
      name: "Bedroom Night Light 2 Light Status"
    battery_level:
      name: "Bedroom Night Light 2 Battery Level"
    illuminance:
      name: "Bedroom Night Light 2 Illuminance"
`

Anything obvious that I missed? I didn't dind anything in the logs. The installation goes smoothly as well. Any help appreciated.

nagyrobi, 2022-10-10:
Is thie related to Bluetooth Proxy? Because that can't be seen in the config.

ptrsconsulting, 2022-10-10:
You're right, it's not. It was late and I didn't notice where I was creating the issue. I'll move it to the correct project and this one can be closed. Sorry for confusion.

closed 2022-10-10 by ptrsconsulting