esphome/bluetooth-proxies #43: olimex-bluetooth-proxy connection issues

marc-gist, 2022-10-10:
I get CONSTANT disconnect, connect on the proxy.  Network is fine, and happens to various HA instances I have tested it on.

[image]

Nothing in the logs except:
```
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
```

Logs also contantly disconnect and reconnect.
```
INFO 10.20.1.31: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO 10.20.1.31: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for 10.20.1.31
```

Both static and dynamic ips tend to do this :(

nagyrobi, 2022-10-10:
Try with `active: false` for both `esp32_ble_tracker scan_parameters` and `bluetooth_proxy`.

marc-gist, 2022-10-10:
> Try with `active: false` for both `esp32_ble_tracker scan_parameters` and `bluetooth_proxy`.

Doesn't the defeat the purpose of this device? That's all it supposed to do is scan and be a proxy.
What's the point in this test?

Thanks

tomi19921702, 2022-10-11:
Got the same problem tried 3 different boards, changed yaml template with a few different settings Removed Esp board from the Device list.

marc-gist, 2022-10-11:
> Got the same problem tried 3 different boards, changed yaml template with a few different settings Removed Esp board from the Device list.

So you fixed it or gave up? 🙂

nagyrobi, 2022-10-11:
@marc-gist it depends on your use case. If you're just gathering sensor data passively you don't need active mode at all.
What device are you using it with?

nagyrobi, 2022-10-11:
```
Please add the following sensors to your config and re-flash the board:
sensor:
- platform: template
  name: "Free Heap Memory"
  lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  unit_of_measurement: "b"
  update_interval: 60s
- platform: uptime
  name: "Uptime"
```
And see if memory consumption stays constant or not, also check uptimes, does it reboot itself often?

tomi19921702, 2022-10-11:
> > Got the same problem tried 3 different boards, changed yaml template with a few different settings Removed Esp board from the Device list.
> 
> So you fixed it or gave up? 🙂

No didnt got it fixed yet, I got mainly switchbot devices and now a couple Inkbird devices thats why I wanted to switch over to the Bluetooth proxy, I trialed before the switchbot mqtt and didnt had any problems.
I will try @nagyrobi Code for the memory consumption and have a look.

tomi19921702, 2022-10-11:
@nagyrobi end of the log it discconnected 
[logs_esp32-bluetooth-proxy-e30150_upload (2).txt](https://github.com/esphome/bluetooth-proxies/files/9753215/logs_esp32-bluetooth-proxy-e30150_upload.2.txt)

nagyrobi, 2022-10-11:
This looks like a memory leak in one of the Bluetooth components, as I also experienced: https://discord.com/channels/429907082951524364/524177279270780928/1029111888967127040

It's a bug, it's being looked at.

tomi19921702, 2022-10-11:
> This looks like a memory leak in one of the Bluetooth components, as I also experienced: https://discord.com/channels/429907082951524364/524177279270780928/1029111888967127040
> 
> It's a bug, it's being looked at. Try setting the logger to `level: INFO` to save MCU resources.

Cheers Mate, which Discord Server is that the Link doesnt work for me.
Thanks.

nagyrobi, 2022-10-11:
ESPHome support Discord server. Such issues are discussed there.