esphome/bluetooth-proxies #44: GL-S10 OTA Safe Boot issue

DJBenson, 2022-10-13:
Following on from #39, even if safe boot is used, I cannot update the GL-S10 using OTA.

I have a revision 2.1 board. I have flashed the initial firmware using the webflash tool and now trying to adopt the device into ESPHome and do the initial OTA. The process will not get beyond single-digit percentage without timing out.

Steps to reproduce:
Flash with intitial binary
Attempt to adopt into ESPHome
OTA times out
Repeat using the safe boot button in HA
OTA times out

INFO Uploading /data/gl-s10-bt-proxy-9ad9b0/.pioenvs/gl-s10-bt-proxy-9ad9b0/firmware.bin (1493552 bytes)
Uploading: [===                                                         ] 5% 
ERROR Error sending data: timed out

I've seen OTA fail on other ESPHome projects due to the flash size being too high to allow the OTA binary - do these stats look ok in that respect;

RAM:   [==        ]  16.7% (used 54592 bytes from 327680 bytes)
Flash: [========  ]  81.4% (used 1493450 bytes from 1835008 bytes)

nagyrobi, 2022-10-13:
Please close this issue and post your findings in this thread: https://github.com/esphome/issues/issues/3682

DJBenson, 2022-10-13:
Always nice if people cross reference their tickets.

closed 2022-10-13 by DJBenson