esphome/bluetooth-proxies #48: Config file for different revisions of GL-S10

Dinth, 2022-11-04:
Hi. Apologies if this is not a correct place for reporting such thing, but shouldnt GL-S10 have different config files for different revisions of board?
Currently im experiencing a problem with GL-S10 v1.0c - i am flashing it with modified config file and it works fine, but as soon as im trying to adopt it to ESPHome instance, it automatically gets reflashed with a config from the repo, which is unmodified and doesnt work with rev v1.0 - therefor after adopting the device it stops working. 
Having two separate config files would resolve this issue

nagyrobi, 2022-11-04:
Solving various issues with GL-S10 is still work in progress. Meanwhile, you should not adopt it anymore from the web interface, instead flash it with your own working config and keep it up to date from ESPHome Dashboard.

balloob, 2022-11-04:
We only want to support the current revision that is shipping. Not anything prior to that. If you change the config file to make it work with your device, you should remove the link with the online package to avoid it getting updated to a version that is incompatible.

closed 2022-11-04 by balloob

Dinth, 2022-11-04:
Thanks for the advice, i will try to remove that line.
This being said, v1.0 is still shipping, i placed my order on 15th October and received 1.0c

nepozs, 2022-11-05:
> We only want to support the current revision that is shipping
I've bought HW 1.0 few days ago (31.10.2022) so… it is still shipping

Dinth, 2022-11-05:
Hey @nepozs. If flashing via a website works for you, you can try here https://blakadder.github.io/bluetooth-proxies/.