esphome/bluetooth-proxies #49: Webpage and firmware for GL.iNet GL-S10 rev2.3 should be separate from GL.iNet GL-S10 rev1.0

nepozs, 2022-11-05:
as it it bricks GL.iNet GL-S10 rev1.0 preflashed with old versions of firmware and it is impossible to use ready made firmware with HW 1.0.

Preflashed versions of HW 1.0 are sometimes available in web shops (I've found such in Poland) and simple Adoption and just Update in ESPHome bricks device…). Of course it is possible to recover with UART flash, but I think that current solution breaks idea of ESPHome Bluetooth Proxies.

nagyrobi, 2022-11-05:
See https://github.com/esphome/bluetooth-proxies/issues/48

nepozs, 2022-11-05:
So website should contain appropriate information: DON"T USE with HW 1.0, but there is one additional problem - there is no possibility to know hardware version before opening device…
[image]

closed 2023-01-05 by nagyrobi

nagyrobi, 2023-01-05:
It has been declared that the site only reflects the latest model available.