esphome/bluetooth-proxies #54: Error attempting to update existing ESPHome bluetooth proxies

MakerDockio, 2022-12-08:
I have just attempted to update my two existing ESPHome esp32 bluetooth proxies via the ESPHome Dashboard in Home Assistant to the latest firmware. When attempting I get the following error in the log.

[image]

The proxies are running on thr Generic ESP32 WROOM modules, with Home Assistant (2022-12) / ESPHome (2022.11.5 ) running on Raspberry Pi 4.

Any pointers would be appreciated.

MakerDockio, 2022-12-08:
Roger that 👍! Cheers Paulus

bbostock, 2022-12-14:
I have also been unable to compile the new update for bluetooth proxies following installation of 2022.12.5 & updating ESPHome. 

After reading this thread I tried the beta, but sadly it fails with the same error: 

`Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/atom-bluetooth-proxy-a6d1e8/.pioenvs/atom-bluetooth-proxy-a6d1e8/CMakeFiles/CMakeOutput.log".
See also "/data/atom-bluetooth-proxy-a6d1e8/.pioenvs/atom-bluetooth-proxy-a6d1e8/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/atom-bluetooth-proxy-a6d1e8/.pioenvs/atom-bluetooth-proxy-a6d1e8/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_8e143 && [1/2] Building C object CMakeFiles/cmTC_8e143.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_8e143.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_8e143.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc: not found
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:296 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 252.15 seconds =========================`

Any help greatly appreciated.

nagyrobi, 2022-12-14:
Try a Clean Build Files from the 3-dot menu!

bbostock, 2022-12-14:
That was the first thing I did. But sadly made no difference.

stefanthoss, 2022-12-15:
I faced similar build issues on my WT32-ETH01 and they were resolved with the ESPHome [release 2022.12.0](https://esphome.io/changelog/2022.12.0.html) from today.

nagyrobi, 2022-12-15:
**xtensa-esp32-elf-g++ not found**
With the recent update of ESPHome, it seems that if you are using a 32-bit OS on a 64-bit RPi4 the toolchain is incorrectly downloaded and you get an error during compilation: `xtensa-esp32-elf-g++ not found`.
We are looking into a fix, but the best recommendation right now is to reinstall your OS to actually use a 64-bit based OS as intended. 
You can rollback ESPHome to before 2022.12.0 but this is not a permanent solution.

bbostock, 2022-12-15:
Ok, thank you for the diagnosis. I'll consider my options before proceeding.

It's very cold here at the moment and I don't think my wife would appreciate me experimenting when the current setup runs our heating system. 
🙏

Gferretta, 2022-12-28:
Hi
I have exactly same error running ESPHOME in a RPi3+ with an SD
No can update existing BT Proxies ESP32 (Generic) over wifi or cable.
No compile at all.

ESPHome 2022.12.3
Home Assistant 2022.12.8
Supervisor 2022.12.1
Operating System 9.4
Frontend 20221213.1 - latest

balloob, 2022-12-30:
The original issue here has been resolved because the new ESPHome has been released.

closed 2022-12-30 by balloob

bbostock, 2022-12-30:
Do I assume that it will be distributed with the January 23 HA release, or is there a way to install ESPHome sooner? (I tried re-installing the ESPHome beta, but it fails too).
Thank you
Bill

bbostock, 2023-01-27:
Despite this apparently being resolved and I'm running the latest HA / ESPHome, it still fails with the same error. 
Please provide some further guidance if I need to do something else.
Thank you
Bill