esphome/bluetooth-proxies #59: bluetooth proxies update fail - compiler error?

crackha, 2023-01-06:
first time poster so if I should be doing this elsewhere please let me know - have been using bluetooth proxies in Home Assistant since December (M5 Atom Lites). I've updated the Home Assistant core to the latest version 2023.1.0 this is now generating a warning saying I need to update the proxies to ESPHome 2022.12.0 or later, but when I attempt to do this via the ESPHome add-on's Web UI, the update fails. Any tips on how to get the updates to succeed?
[image]
 Thanks in advance. cheers

balloob, 2023-01-06:
Make sure your ESPHome add-on is up to date.

crackha, 2023-01-06:
Thanks for the comment - I uninstalled it and then reinstalled it today, and the current version installed is 2022.12.3. Appreciate all tips! cheers

crackha, 2023-01-06:
Just tried to update again via HA's ESPHome add-on's Web UI and the picture shows what happens - thanks again
[image]

crackha, 2023-01-06:
any other thoughts/tips? cheers

crackha, 2023-01-06:
downloading the diagnostics from the HA integration page, shows that the proxies themselves are running the older version (why they need to update I guess) - another punter thought that it was a compiler issue - either way it's beyond my skill - the best bit of the proxy inclusion into HA was the ability for novice's like me to use them - hopefully this can be fixed? all help appreciated - cheers

crackha, 2023-01-06:
here is the diagnostic bit I was referring to:
  "uses_password": false,
        "name": "atom-bluetooth-proxy-06d720",
        "mac_address": "**REDACTED**",
        "compilation_time": "Dec 13 2022, 06:54:17",
        "model": "m5stack-atom",
        "manufacturer": "Espressif",
        "has_deep_sleep": false,
        "esphome_version": "2022.12.0b5",
        "project_name": "esphome.bluetooth-proxy",
        "project_version": "1.0",
        "webserver_port": 0,
        "bluetooth_proxy_version": 3

nagyrobi, 2023-01-06:
This repository is only for hosting the [bluetooth-proxies](https://esphome.github.io/bluetooth-proxies/) web-installer site. Please only post here issues related to the installer functionality, or the website itself.

For **bug reports** related to `bluetooth_proxy` component and ESPHome issues, please go to [https://github.com/esphome/issues](https://github.com/esphome/issues/issues)

For **feature requests** related to `bluetooth_proxy` component, go to [https://github.com/esphome/feature-requests](https://github.com/esphome/feature-requests/issues)

For fast answers to **direct questions**, _such as this one_, head on to our [Discord](https://discord.gg/KhAMKrd) support channel.

Thank you!

closed 2023-01-06 by nagyrobi

crackha, 2023-01-07:
Thank you @nagyrobi for the guidance - I am def new at this and apologise if I am putting comments in the incorrect spots. Regarding the bluetooth-proxies web-installer site (http://esphome.github.io/bluetooth-proxies/), with the recent updates to the Bluetooth Proxy setup in HA (see https://esphome.io/changelog/2022.12.0.html), and with the hope that minimal programming novices like myself can use these awesome features, may I humbly request that the bluetooth-proxies web-installer site (http://esphome.github.io/bluetooth-proxies/) is updated to have the latest firmware loaded at the time of setting up the Atom Lite (and other units), with future updates occurring via WiFi? Thanks for reading, and I hope this was an appropriate post about the installer functionality? Cheers