esphome/bluetooth-proxies #60: GL.iNet GL-S10 - No Ethernet connection with hardware v2.1?

JohnyBeGood, 2023-01-06:
Hey all,

I followed [The Best Home Assistant Bluetooth Proxy: GL.iNet GL-S10](https://blakadder.com/gl-s10/) tutorial and flashed it using [official web installer](https://esphome.github.io/bluetooth-proxies/?diy) but it never connects to my POE network nor using S10 GLI.net Android app. 
I'm unable to find anything online and of course, Home Assistant does not see it.
Any help would be appreciated!

Thanks in advance!

oywino, 2023-01-07:
I have the same problem.
My GL-S10 is also v2.1 and it was working fine until a couple of days back (after updating HA to 2023.1) I got the following message in HA:  

`To improve Bluetooth reliability and performance, we highly recommend updating gl-s10-bt-proxy-1e66cc with ESPHome 2022.12.0 or later. When updating the device to ESPHome 2022.12.0, it is recommended to use a serial cable instead of an over-the-air update to take advantage of the new partition scheme.`

So, I updated the device using the same serial cable and procedure (https://esphome.github.io/bluetooth-proxies/?diy) as I used when I first bought the unit. The update was reported successful, but now it won't come online at all.

So, I guess from having a fairly unstable Ethernet driver, we now have  version with total failure Ethernet driver. 😢

Is there any way to downgrade to a previous firmware?

JohnyBeGood, 2023-01-08:
Thanks for the reply!

I'm glad it's not just me! In my case, it never showed up in HA as the tutorial shows. How you were able to add it to the HA?
I have no idea where to start with troubleshooting.

oywino, 2023-01-08:
I did manage to get it online today. Here's what I did:
I first installed ESPHome in a separate docker container on my NAS using `esphome/esphome:latest`
Then I connected the GL-S10 using a USB port on my Windows PC while running ESPHome Web version.
Then, from ESPHome I manually downloaded the firmware (which involved a script that compiled and linked 100+ files) before finally creating the *.bin file i've included here.
Then I manually installed that file o
[gl-s10-bt-proxy-1e66cc-factory.zip](https://github.com/esphome/bluetooth-proxies/files/10369585/gl-s10-bt-proxy-1e66cc-factory.zip)
nto the GL-S10 from the menu in ESPHome.
Then I toggled the Ethernet line speed between Auto and 10MF and it came online.

Regretfully, it is still suffering from a massive package loss, so it isn't very useful for any practical purpose, but at least it's back on line.  It shows up in Home Assistant and the firmware is now: 2022.12.3
I do not know if this is the most recent version, and I still don't understand why it suffers from such heavy packet losses.

oywino, 2023-01-08:
Reading the YAML file at `github://esphome/bluetooth-proxies/gl-s10.yaml@main` I see that it is configured for the GL-S10 with revision board v2.3 and it says to comment out if you want to change to v1.0
Now, there is no mention of v2.1 boards which we have, but the said section in the YAML file relates directly to the Ethernet driver configuration.  That lead me to believe that it would perhaps be worth while to test the v1.0 config, but I do not know how to do that.  What I did was this:
I copied the YAML script from GitHUB and modified it as suggested. Then I clicked `EDIT` in ESPHome and pasted the YAML over the short version that was already in there, and selected `INSTALL` 
That produced a bin file named `gl-s10.bt-proxy-factory.bin` which I then flased onto the board.

The result was disappointing. I never could get it online, so I went back to the original file again.  But I may have done something wrong. The Web `https://esphome.github.io/bluetooth-proxies/?diy` has no instruction on how to choose/select or modify the YAML file, so I have no control of which file is being used.

Ashthos, 2023-01-15:
I also had issues with my V2.1 board. 

The firmware provided above by @oywino unfortunately has the wrong checksum for my board, luckily I had a old firmware from a previous patching.

If the one above doesn't work for you, try this one:

[gl-s10-bt-proxy.zip](https://github.com/esphome/bluetooth-proxies/files/10420231/gl-s10-bt-proxy.zip)

Firmware: 2022.10.0, ESPHome version.

Looks like the flasher needs to be updated to allow selection of the hardware for the Gi.net boards.

oywino, 2023-01-15:
> If the one above doesn't work for you, try this one:
> 
> [gl-s10-bt-proxy.zip](https://github.com/esphome/bluetooth-proxies/files/10420231/gl-s10-bt-proxy.zip)
> 
> Firmware: 2022.10.0, ESPHome version.
> 
> Looks like the flasher needs to be updated to allow selection of the hardware for the Gi.net boards.

Thanks so much for stepping in. But how do I flash with the proposed version above, when the flasher lacks the option to select which file to flash ?

balloob, 2023-01-15:
Use https://web.esphome.io to install any binary.

oywino, 2023-01-15:
@balloob - twice I tried to flash the .bin file you provided. Both attempts reported successful flashing, but my GL-S10 refused to boot on both attempts.  So I reflashed it again with default method offered at [https://esphome.github.io/bluetooth-proxies/?diy](url) it boots up ok, but I'm back to square one with an Ethernet connection that suffers from massive package losses.

Did I do something wrong? I extracted the file from your ZIP and selected it using [https://web.esphome.io/](url) and flashed it.
As you say, there is no way to select the HW (device) so perhaps that's causing a problem. The LED on my serial board was blinking very differently when I use the [https://esphome.github.io/bluetooth-proxies/?diy](url) method because there I can select the HW, but not the file i'd like to use.

But what about the YAML ?  should the defaul YAML file also be replaced with this one before flashing:

`# Instructions on opening and wiring for flashing on https://blakadder.com/gl-s10

substitutions:
  name: gl-s10-bt-proxy

esphome:
  name: ${name}
  name_add_mac_suffix: true
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"
# turn on Power LED when esphome boots
  on_boot:
    then:
      - output.turn_on: power_led

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

# Configuration fo V2.3 hardware revision
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO5

# Comment the above and use this instead for V1.0 revision of the hardware
# ethernet:
#   type: LAN8720
#   mdc_pin: GPIO23
#   mdio_pin: GPIO18
#   clk_mode: GPIO17_OUT
#   phy_addr: 1

api:
logger:
ota:

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/gl-s10.yaml@main

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
# Bluetooth LED blinks when receiving Bluetooth advertising
  on_ble_advertise:
    then:
      - output.turn_on: bluetooth_led
      - delay: 0.5s
      - output.turn_off: bluetooth_led

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic

## DEVICE SPECIFIC CONFIGURATION
# network LED (white one) configured as status led
status_led:
  pin:
    number: GPIO32
    inverted: true

# button on the side labeled RESET
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO33
      inverted: true
    name: "Reset Button"

# output settings for LED's marked Power and Bluetooth
# power LED use: see code line 12
# bluetooth LED use: see code line 41
output:
  - platform: gpio
    pin: GPIO14
    inverted: true
    id: power_led
  - platform: gpio
    pin: GPIO12
    inverted: true
    id: bluetooth_led

# since these pins are broken out inside and labeled as I2C pins they're configured here
i2c:
  sda: 15
  scl: 13
  scan: true`

Ashthos, 2023-01-15:
@oywino The firmware I uploaded above was the one I re-flashed back onto my GL s10 when the esphome web flasher borked my board (claims successful flash, but no activity from the s10 - as per this thread..).  After re-flashing the above firmware my s10 connected back to HA fine and has been working as expected (which apart from claiming it is rebooting all the time in the logs, has appeared to be reasonably reliable).

When I first got my s10, I initially had issues flashing the s10 (due to a dodgy power supply). As part of my process to get it to flash successfully, I created that firmware and uploaded it using the esp flasher windows app (https://github.com/esphome/esphome-flasher/releases).

Unfortunately, I am no expert on this stuff, but I just provided my firmware in case it helps someone else. Sorry, I can't be of more help.

oywino, 2023-01-18:
@Ashtos - I tried again to flash the GL-S10 using the Windows Flasher and the .bin file you suggested, and remarkably - this time the device does boot up as it should. It failed to boot when I used https://web.esphome.io/ and the same .bin file, although the flashing process reported successful in both attempts.  Why one method produced a working version, while the other did not - beats me.
Anyway, the .bin file you offered is Firmware: 2022.10.0, ESPHome version (as you wrote), but Home Assistant immediately complains and tells me to update to 2022.12.0 or later.  In addition, your 2022.10.0 .bin also suffers from package loss, but not as bad as the the others when the Ethernet HUB is set to 10MF.
Did you yourself update to 2022.12.0 ?

Ashthos, 2023-01-20:
@oywino Glad you are back to a 'working' state. 

I flashed the GL-s10 when I first got it (in October) with the bluetooth proxy tools (following Blakadders tutorial). First time I've attempted to change anything was the other day when I, once again, tried using the bluetooth proxy tools to update (after being prompted to do so in HA) - I got the failure everyone else has been seeing, so I rolled back to my old firmware.

Personally I will just live with the packet loss (which doesn't impact me directly as all my bluetooth devices still work flawlessly with the s10 via HA). Maybe in a 'version or two' I'll try the update process again which will include a working ethernet driver?

oywino, 2023-01-20:
@Ashthos  Thanks for your reply.
Following Blackadders tutorial resulted in a Firmware with an unacceptable high packet loss. Consequently, theonly way to update (or downgrade) the firmware is by using USB2Serial board. These ESP based units should be OTA updatable, at least with cabled Ethernet connection.  My understanding is that the HA-Team (company?) which is behind NabuCasa has purchased the ESPHome system and thus it is a common ownership behind HA and ESPHome (?)  I have no idea who @Blackadder is, but I find it unbelievable that the HA Team seems unable (or unwilling?) to clean up the core ESP-32 driver and iron out these fundamental bugs?
When I installed the .bin file you offered, the packet losses disappeared completely, but in return HA complains constantly about the need to upgrade. It makes me nervous not being able to update a product like this. It's like waiting for a disaster; One day a new HA release may cause the BT-Proxy to fail completely unless updated.
After flashing your .bin file, I tried installing 12.3.0 using the YAML file listed above, but the result is still massive packet losses.
I must admit - I do not fully understand the purpose of the YAML file. How did you manage to "compile" a working .bin file?

oywino, 2023-01-29:
I tried @blackadder's upgrade path for my GL-S10 again today, and the firmware was upgraded to 2022.12.6 But checking on GitHUB, the latest version should be 2022.12.8 [https://github.com/esphome/esphome/releases/](url)
Why @blackadder's upgrade isn't updated, I do not know. However, when I checked the device log, it strangely enough says that the Ethernet connection is only Half-Duplex (?)
```
[01:06:24][C][ethernet:163]: Ethernet:
[01:06:24][C][ethernet:281]:   IP Address: 192.168.1.160
[01:06:24][C][ethernet:282]:   Hostname: 'gl-s10-bt-proxy-1e66cc'
[01:06:24][C][ethernet:283]:   Subnet: 255.255.255.0
[01:06:24][C][ethernet:284]:   Gateway: 192.168.1.1
[01:06:24][C][ethernet:289]:   DNS1: 192.168.1.1
[01:06:24][C][ethernet:290]:   DNS2: 0.0.0.0
[01:06:24][C][ethernet:297]:   MAC Address: 0C:B8:15:1E:66:CF
[01:06:24][C][ethernet:302]:   Is Full Duplex: NO
[01:06:24][C][ethernet:307]:   Link Speed: 10
[01:06:24][C][ethernet:166]:   Power Pin: 5
[01:06:24][C][ethernet:168]:   MDC Pin: 23
[01:06:24][C][ethernet:169]:   MDIO Pin: 18
[01:06:24][C][ethernet:170]:   Type: IP101
```
If that is in fact the case, no wonder it suffers from high packet loss.
Is it possible to correct this somewhere in the YAML file?

JohnyBeGood, 2023-01-29:
Thank you all for your replies! Even after I tried updating 2 days ago using https://web.esphome.io/ when I connect it to the ethernet it never connects to my network and HA does not see it. Can someone please list the steps they took for connecting to the ethernet?
So far it's useless and has no use.

oywino, 2023-01-29:
@jesserockz Could you please step in and help us out?
It seems that nobody is able to get the GL-S10 to connect via LAN without massive package losses.
I have successfully flashed 2022.12.8 but to no avail.
I'm using this YAML :  [https://github.com/esphome/bluetooth-proxies/blob/main/gl-s10.yaml](url)

nareddyt, 2023-02-18:
How are you guys measuring packet loss?

I have GL-S10 v2.1 and had no issues getting it to show up in ESPHome dashboard. It is mostly working, but sometimes I see it go unavailable. Happens maybe twice a day.

nareddyt, 2023-02-18:
Related discussion: https://community.home-assistant.io/t/gl-s10-v2-1-packet-loss-issues/486575

jr-air, 2023-03-24:
Did some quick testing while referencing the ESPHome Ethernet Componet doc, and now I seem to have a much more "stable" Ethernet connection

I changed
clk_mode: GPIO17_OUT
-to-
clk_mode: GPIO0_IN

For the IP101, the only example on this page showed using GPIO0 instead of GPIO17.
https://esphome.io/components/ethernet.html

Can the respective "[gl-s10.yaml]" file be updated, as I have to manually copy the yaml over then modify, instead of pulling direct from github.

Also, OTA works without failing mid-way.

JohnyBeGood, 2023-03-24:
> Did some quick testing while referencing the ESPHome Ethernet Componet doc, and now I seem to have a much more "stable" Ethernet connection
> 
> I changed clk_mode: GPIO17_OUT -to- clk_mode: GPIO0_IN
> 
> For the IP101, the only example on this page showed using GPIO0 instead of GPIO17. https://esphome.io/components/ethernet.html
> 
> Can the respective "[gl-s10.yaml]" file be updated, as I have to manually copy the yaml over then modify, instead of pulling direct from github.
> 
> Also, OTA works without failing mid-way.
> 
> (My GL-S10 has a board rev of v2.1)

Thanks for taking the time to reply! 
Mine GL-S10 is still sitting and collecting dust. Can you please list the steps on how you flashed it?
I can flash it by going to https://esphome.github.io/bluetooth-proxies/?diy but when I connect it to POE ethernet it doesn’t seem to connect.

nareddyt, 2023-03-24:
Great find @jr-air! I can volunteer to test this out over the weekend on my devices, and then update the docs if it works.

Go0oSer, 2023-03-24:
I updated my unit. Ill pass along my feedback after letting it bake for a little bit. Thank you for sharing your solution.

jr-air, 2023-03-25:
> Thanks for taking the time to reply! Mine GL-S10 is still sitting and collecting dust. Can you please list the steps on how you flashed it? I can flash it by going to https://esphome.github.io/bluetooth-proxies/?diy but when I connect it to POE ethernet it doesn’t seem to connect.

If using the listed flash site, it will not work until the associated yaml is updated in this repo. 

I copied the yaml to text file, modified it so it wouldn't pull from the repo (not ideal), edited the one line mentioned above, then flashed it thru my HA ESPHOME install (via UART pins).

I bought four of these things and now have three of four flashed with the fix. I ended up burning one up, since I didn't know you would damage the board if power by POE while connected to the UART pins (yes USB was not connected as doing POE and USB at same time will also fry it). So "always" use USB power when flashing using the UART! 

I ran OTA over ethernet on all three successfully.

JohnyBeGood, 2023-03-25:
> If using the listed flash site, it will not work until the associated yaml is updated in this repo.
>
> I copied the yaml to text file, modified it so it wouldn't pull from the repo (not ideal), edited [...]

Thank you so much for taking the time to reply!
I'm away from home currently but will try your suggestion as soon as I get a chance.

Yes, I read https://blakadder.com/gl-s10/ not to connect both, but after countless tries of various combinations between flashing with the USB and trying POE I'm very surprised I did not fry my unit :)

oywino, 2023-03-25:
Hi guys,
Successful OTA update **once** after your YAML modification may work, but will it not then pull a version without your mod, and hence set you back to square one again? How do you secure that the correct (modified) YAML is used during consequtive OTA's?

A second question I have, is this: Does it work as a BLE Proxy after OTA with your mod? How did you test it?

jr-air, 2023-03-25:
oywino, I strongly recommend waiting till the yaml file in this repo gets updated (after a few test) as this will make everything so much easier. But as unit is mentioned also a DIY specifically, I'll post the manual method I used.

One quick note, OTA only applies "after" you get the initial firmware on the device. This was the man issue I was addressing as the ethernet was unreliable per this thread. The BLE functionality should not have changed or been affected.

1) You must have ESPHOME installed. Am assuming as part of your Home Assistant or Start one vis docker (more advanced method)
2) Copy the "gl-s10.yaml" file to a text editor.
3) Don't use the existing Web Installer for this device (at least not until the ethernet change is tested then updated)
4) Edit the lines in your text editor (see end of this post)
5) In your local ESPHOME, add new device, choose generic ESP32 board and skip any flashing.
6) Edit the config for the dummy device created, and paste in the modified lines from your text editor (replace all lines)
7) Perform the same flashing steps you would use for the main web installer, but via your local ESPHOME instead
    - Key is the lines removed below, as your firmware will revert to one based off this repo.

gl-s10.yaml edits used just for testing via local ESPHOME install

_remove_:
substitutions:
  name: gl-s10-bt-proxy
  friendly_name: Bluetooth Proxy
dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/gl-s10.yaml@main

_change_:
esphome:
  name: ${name}     - ${name} to "bt-proxy-01" or similar 
  friendly_name: ${friendly_name}     - ${friendly_name} to  "bt-proxy-01" or similar 
  name_add_mac_suffix: true   - true to false or just delete line
ethernet:
  clk_mode: GPIO17_OUT    - GPIO17_OUT to GPIO0_IN

Go0oSer, 2023-03-25:
> oywino, I strongly recommend waiting till the yaml file in this repo gets updated (after a few test) as this will make everything so much easier. But as unit is mentioned as an advance DIY specifically, I'll post the manual method I used.
> 
> One quick note, OTA only applies "after" you get the initial firmware on the device. This was the man issue I was addressing as the ethernet was unreliable per this thread. The BLE functionality should not have changed or been affected.
> 
>     1. You must have ESPHOME installed. Am assuming as part of your Home Assistant or Start one vis docker (more advanced method)
> 
>     2. Copy the "gl-s10.yaml" file to a text editor.
> 
>     3. Don't use the existing Web Installer for this device (at least not until the ethernet change is tested then updated)
> 
>     4. Edit the lines in your text editor (see end of this post)
> 
>     5. In your local ESPHOME, add new device, choose generic ESP32 board and skip any flashing.
> 
>     6. Edit the config for the dummy device created, and paste in the modified lines from your text editor (replace all lines)
> 
>     7. Perform the same flashing steps you would use for the main web installer, but via your local ESPHOME instead
>        
>        * Key is the lines removed below, as your firmware will revert to one based off this repo.
> 
> 
> gl-s10.yaml edits used just for testing via local ESPHOME install
> 
> _remove these lines_:
> 
> ```
> substitutions:
>   name: gl-s10-bt-proxy
>   friendly_name: Bluetooth Proxy
> dashboard_import:
>   package_import_url: github://esphome/bluetooth-proxies/gl-s10.yaml@main
> ```
> 
> _change_:
> 
> ```
> esphome:
>   name: ${name}     # ${name} to "bt-proxy-01" or similar 
>   friendly_name: ${friendly_name}      # ${friendly_name} to  "bt-proxy-01" or similar 
>   name_add_mac_suffix: true   # true to false or just delete line
> ethernet:
>   clk_mode: GPIO17_OUT    # GPIO17_OUT to GPIO0_IN
> ```

PR submitted - https://github.com/esphome/bluetooth-proxies/pull/77

jesserockz, 2023-03-26:
Just a quick note,

You should be able to simply add: 

```yaml
ethernet:
  clk_mode: GPIO0_IN
```

To the bottom of the adopted file that uses the package from GitHub and it should override the pulled version.

Ashthos, 2023-03-26:
> oywino, I strongly recommend waiting till the yaml file in this repo gets updated (after a few test) as this will make everything so much easier. But as unit is mentioned as an advance DIY specifically, I'll post the manual method I used.

Looks like they don't support the v2.1 hardware (only v2.3) therefore manually flashing the device is our only option.

raoulbhatia, 2023-03-26:
FYI I also tested the change with presumably hardware revision 2.1 and can confirm the good results.

[image]

LGTM!

nareddyt, 2023-03-26:
Works great for me on v2.1 of the board.

I see @Go0oSer has a PR https://github.com/esphome/bluetooth-proxies/pull/78 to for.

efaden, 2023-03-27:
Updated mine, not a single dropped packet.

verstrepeng, 2023-03-28:
Just changed to config and installed firmware by wire. Reverted my switchport also back to autodetect.
It connects at 100/Full and seems not to miss a packet :-)

jjross, 2023-03-31:
So this is what I have in my config (downloaded from the website):

`# Configuration fo V2.3 hardware revision
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO5`

Are we saying I would change that to:

`# Configuration fo V2.3 hardware revision
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO5`

HeFeng1947, 2023-03-31:
Hi guys, please refer here #79

jesserockz, 2023-04-05:
The published firmware has been updated (#77) with `GPIO0_IN` to match the information from #79

closed 2023-04-05 by jesserockz

JohnyBeGood, 2023-04-13:
Even with the new flash using https://esphome.github.io/bluetooth-proxies/?diy and 0 pocket loss, I'm still unable to use it.
After the flash while connected via POE in HA gets discovered, but it shows as unavailable (screenshot). In ESPHome it does not appear. How did you guys get it work?

[image]
[image]

jjross, 2023-04-13:
Mine just seems to work.  HA discovered it and added it no problem.

oywino, 2023-04-13:
How do you power it?
What version of ESPHome do you use?
Where/how is ESPHome hosted/running?

JohnyBeGood, 2023-04-13:
> Mine just seems to work. HA discovered it and added it no problem.
> […](#)

Wow, so in your case it automatically show up in ESPHome?

JohnyBeGood, 2023-04-13:
> How do you power it? What version of ESPHome do you use? Where/how is ESPHome hosted/running?

1. It's powered over ethernet (POE)
2. ESPHome on HA Current version: 2023.3.2
3. On HA supervisor in VM

oywino, 2023-04-13:
Can  you try to power it separately over USB?
Can you see the "familiar" LED flashing between white&green just after power on, before it stabilizes on flashing green (link light)?
ESPHome stores a copy of all the build files as well as the YAML config in a local /config folder.  I'm not sure where exactly this folder can be found in a HA Supervised VM, but in my experience, finding this folder and deleting all it's content and then power-cycle the GS-10 has helped me get it online after a version upgrade.

JohnyBeGood, 2023-04-16:
> Can you try to power it separately over USB? Can you see the "familiar" LED flashing between white&green just after power on, before it stabilizes on flashing green (link light)? ESPHome stores a [...]

Thanks for the reply!
When flashed Wi-Fi service is turned off so only POE works and I can't power it with both. It would fry it. Once plugged in POE I do get familiar blinking lights. Seems like it's working, but not according to Home Assistant dashboard.