esphome/bluetooth-proxies #68: Why doesn't the website upgrade the ESP32 to the latest released version?

Gferretta, 2023-02-21:
Hi

I am one of the many who have problems upgrading the version via OTA due to errors in the compilation on an RPi3b+.
But I don't understand why if I update my generic ESP32 from https://esphome.github.io/bluetooth-proxies/ to this day it continues to install version 2022.12.8 when version 2023.2.3 is released
The website does the complete deletion and the new installation over my ESP32 without errors, it connects perfectly to the WiFi, but with version 2022.12.8, so in my ESPHome I still get the upgrade warning for my devices, since I have the version installed 2023.2.3
Any ideas?

Thank you
Regards

balloob, 2023-02-21:
Fixed in https://github.com/esphome/bluetooth-proxies/pull/69

closed 2023-02-21 by balloob

Gferretta, 2023-02-22:
> Fixed in #69

Is the update already available on the web or do I have to wait until next Monday?
Thank you

Gferretta, 2023-02-28:
Hi
Today (monday), ending the day, an old firmware version is still installing the website.
V2023.2.3 When the current one is 2023.2.4
Wouldn't it be better than for each release of a new version of ESPHOME is automatically compile and that the website always install the last current version?
Wait since I can't update via ota
Thank you

[image]

Gferretta, 2023-03-08:
Today 3/8/23 the website is still installing V2023.2.3 when the current one is 2023.2.4
what's going on?
Why don't you fix to install the latest version available on ESPHome?
As I mentioned before, the compilation from ESPHome produces an error and I cannot do it via OTA.
This was already reported by many other users and we have not had an answer to that problem either.
Thank you
Regards

raoulbhatia, 2023-03-09:
It seems the workflow kicked in and the build process worked fine, see https://github.com/esphome/bluetooth-proxies/actions/runs/4339889757

However, the `consolidate ` and `upload` jobs didn't run.
I assume the related if statements need to be double checked, see https://github.com/esphome/bluetooth-proxies/actions/runs/4339889757/workflow#L61

Raoul

Gferretta, 2023-03-09:
> It seems the workflow kicked in and the build process worked fine, see https://github.com/esphome/bluetooth-proxies/actions/runs/4339889757
>
> However, the `consolidate ` and `upload` jobs didn't run. [...]

Exactly
Thanks!!