esphome/bluetooth-proxies #73: What are the step to start from Scratch?

howchik, 2023-03-04:
Hello, 
I am new to this. spend all day trying to start somewhere, but not good tutorial or instructions. 

How to use YAML? 
Is it for ArdruinoIDE or PlatformIO? 
.INI file should have code for the device. What do you get it? 

Literally spent 5 hours, and still not progress. 

Anything helps.

Thank you,

balloob, 2023-03-04:
I have no idea what you're talking about. This issue tracker is for issues about ready-made firmware for Bluetooth Proxies.

closed 2023-03-04 by balloob

howchik, 2023-03-04:
I want to use bluetooth proxies. How to use it? Any instructions? 

I am new to the space. 

seems like i need to use"[esp32-generic.yaml](https://github.com/esphome/bluetooth-proxies/blob/main/esp32-generic.yaml)"
but where and how?

raoulbhatia, 2023-03-04:
@howchik a real life usage example for `GL.iNet GL-S10` can be found here https://blakadder.com/gl-s10/ .
Please note that there are some issues and i think this is the reason it is no longer listed at https://esphome.github.io/bluetooth-proxies/

If you want to build/adapt the configuration yourself, you can check the `yaml` files in this git repository as well as the documentation here https://esphome.io/components/bluetooth_proxy.html

In short:
1. Install `esphome` - either directly, or pull a docker container.
2. Copy the `yaml` file from this repository and save i.e. as `your_device.yaml`.
3. Adjust it to your environment (i.e. passwords, re-configure to pure WiFi if you want to, etc.).
4. Use `esphome run your_device.yaml` (direct install) or `docker run -it --rm -v `pwd`:/config esphome/esphome:latest run /config/your_device.yaml` [1] (Docker version untested by me)

[1] Adapted from https://github.com/esphome/issues/issues/3682#issuecomment-1441373821 thanks @jsimonetti

howchik, 2023-03-05:
@raoulbhatia Thank you very much for the instruction! Greatly appriciate your time! 

Will follow your examples!