esphome/bluetooth-proxies #81: With latest update GL.iNet GL-S10 still does not work in HA?

JohnyBeGood, 2023-04-13:
Even with the new flash using https://esphome.github.io/bluetooth-proxies/?diy and 0 pocket loss, I'm still unable to use it.
After the flash while connected via POE in HA gets discovered, but it shows as unavailable (screenshot). In ESPHome it does not appear. How did you guys get it work?

[image]
[image]

bdraco, 2023-08-06:
There was a problem with the LED blinker, causing all the advertisements to be decoded

Can you try reflashing with the latest ?

JohnyBeGood, 2023-08-08:
Thanks for the reply!

I did flash it again using this link and selecting GL-iNet https://esphome.github.io/bluetooth-proxies/?diy and it doesn't seems like it's working properly?

logs shows this:

```
Logger: aioesphomeapi.reconnect_logic
Source: runner.py:179
First occurred: 5:45:44 PM (1 occurrences)
Last logged: 5:45:44 PM

Can't connect to ESPHome API for gl-s10-bt-proxy-af1d2c @ 192.168.1.240: Error connecting to ('192.168.1.240', 6053): [Errno 113] Connect call failed ('192.168.1.240', 6053) (SocketAPIError)
```

```
Logger: aioesphomeapi.connection
Source: runner.py:179
First occurred: 5:45:41 PM (3 occurrences)
Last logged: 6:21:48 PM

gl-s10-bt-proxy-af1d2c @ 192.168.1.240: Connection error occurred: Ping response not received after 90.0 seconds
gl-s10-bt-proxy-af1d2c @ 192.168.1.240: Connection error occurred: [Errno 104] Connection reset by peer
```

[image]

[image]

bdraco, 2023-08-08:
Looks like it's crashing. You'll need to attach a serial / usb cable and get logs

JohnyBeGood, 2023-08-08:
> Looks like it's crashing. You'll need to attach a serial / usb cable and get logs

Here's the log. It just sits at this line: `I (0) cpu_start: Starting scheduler on APP CPU.`

What could it be?

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14944
ho 0 tail 12 room 4
load:0x40080400,len:3816
entry 0x40080698
I (29) boot: ESP-IDF 4.4.4 2nd stage bootloader
I (29) boot: compile time 04:17:20
I (29) boot: chip revision: v1.0
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (76) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (84) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (91) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (99) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (106) boot: End of partition table
I (111) boot_comm: chip revision: 1, min. application chip revision: 0
I (118) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2c3b8h (181176) map
I (192) esp_image: segment 1: paddr=0003c3e0 vaddr=3ffbdb60 size=03c38h ( 15416) load
I (198) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=b471ch (739100) map
I (466) esp_image: segment 3: paddr=000f4744 vaddr=3ffc1798 size=012e0h (  4832) load
I (468) esp_image: segment 4: paddr=000f5a2c vaddr=40080000 size=18dd4h (101844) load
I (526) boot: Loaded app from partition at offset 0x10000
I (526) boot: Disabling RNG early entropy source...
I (537) cpu_start: Pro cpu up.
I (538) cpu_start: Starting app cpu, entry point is 0x40082758
I (0) cpu_start: App cpu up.
I (554) cpu_start: Pro cpu start user code
I (554) cpu_start: cpu freq: 160000000
I (554) cpu_start: Application information:
I (558) cpu_start: Project name:     gl-s10-bt-proxy
I (564) cpu_start: App version:      2023.7.1
I (569) cpu_start: Compile time:     Aug  7 2023 04:15:11
I (575) cpu_start: ELF file SHA256:  3b4279038d60bb6b...
I (581) cpu_start: ESP-IDF:          4.4.4
I (586) heap_init: Initializing. RAM available for dynamic allocation:
I (593) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (599) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (605) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (612) heap_init: At 3FFCA2C8 len 00015D38 (87 KiB): DRAM
I (618) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (624) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (630) heap_init: At 40098DD4 len 0000722C (28 KiB): IRAM
I (638) spi_flash: detected chip: generic
I (641) spi_flash: flash io: dio
I (647) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

```

JohnyBeGood, 2023-08-22:
Purchased a brand new unit (just to rule out I didn't get bad one the first time) and followed all flashing steps. 
After clicking on adapt in HA it seems like install went fine, but it keeps scanning and it never stops.

Below is the snippet of the log: 

```
Compiling /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/lib4ba/noise-c/protocol/signstate.o
Compiling /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/lib4ba/noise-c/protocol/symmetricstate.o
Compiling /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/lib4ba/noise-c/protocol/util.o
Archiving /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/lib4ba/libnoise-c.a
Archiving /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/lib261/libsodium.a
Linking /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.elf
RAM:   [==        ]  15.1% (used 49492 bytes from 327680 bytes)
Flash: [======    ]  60.5% (used 1110753 bytes from 1835008 bytes)
Building /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.bin"], ["/data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.elf"])
Wrote 0x120950 bytes to file /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 86.64 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of gl-s10-bt-proxy-cfb2bc.local
INFO  -> 192.168.1.245
INFO Uploading /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.bin (1116496 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from gl-s10-bt-proxy-cfb2bc.local using esphome API
INFO Successfully connected to gl-s10-bt-proxy-cfb2bc.local
[17:47:10][I][app:102]: ESPHome version 2023.8.2 compiled on Aug 21 2023, 17:45:54
[17:47:10][I][app:104]: Project esphome.bluetooth-proxy version 1.0
[17:47:10][C][status_led:019]: Status LED:
[17:47:10][C][status_led:020]:   Pin: GPIO32
[17:47:10][C][logger:301]: Logger:
[17:47:10][C][logger:302]:   Level: DEBUG
[17:47:10][C][logger:303]:   Log Baud Rate: 115200
[17:47:10][C][logger:305]:   Hardware UART: UART0
[17:47:10][C][i2c.idf:061]: I2C Bus:
[17:47:10][C][i2c.idf:062]:   SDA Pin: GPIO15
[17:47:10][C][i2c.idf:063]:   SCL Pin: GPIO13
[17:47:10][C][i2c.idf:064]:   Frequency: 50000 Hz
[17:47:10][C][i2c.idf:067]:   Recovery: bus successfully recovered
[17:47:10][I][i2c.idf:077]: Results from i2c bus scan:
[17:47:10][I][i2c.idf:079]: Found no i2c devices!
[17:47:11][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Reset Button'
[17:47:11][C][gpio.binary_sensor:016]:   Pin: GPIO33
[17:47:11][C][gpio.output:010]: GPIO Binary Output:
[17:47:11][C][gpio.output:011]:   Pin: GPIO14
[17:47:11][C][gpio.output:012]:   Inverted: YES
[17:47:11][C][gpio.output:010]: GPIO Binary Output:
[17:47:11][C][gpio.output:011]:   Pin: GPIO12
[17:47:11][C][gpio.output:012]:   Inverted: YES
[17:47:11][C][bluetooth_proxy:088]: Bluetooth Proxy:
[17:47:11][C][bluetooth_proxy:089]:   Active: YES
[17:47:11][C][safe_mode.button:022]: Safe Mode Button 'Safe Mode Boot'
[17:47:11][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[17:47:11][C][esp32_ble:238]: ESP32 BLE:
[17:47:11][C][esp32_ble:240]:   MAC address: 0C:8B:95:CF:B2:XX
[17:47:11][C][esp32_ble:241]:   IO Capability: none
[17:47:11][C][esp32_ble_tracker:617]: BLE Tracker:
[17:47:11][C][esp32_ble_tracker:618]:   Scan Duration: 300 s
[17:47:11][C][esp32_ble_tracker:619]:   Scan Interval: 1100.0 ms
[17:47:11][C][esp32_ble_tracker:620]:   Scan Window: 1100.0 ms
[17:47:11][C][esp32_ble_tracker:621]:   Scan Type: ACTIVE
[17:47:11][C][esp32_ble_tracker:622]:   Continuous Scanning: True
[17:47:11][C][ethernet:221]: Ethernet:
[17:47:11][C][ethernet:363]:   IP Address: 192.168.1.245
[17:47:11][C][ethernet:364]:   Hostname: 'gl-s10-bt-proxy-cfb2bc'
[17:47:11][C][ethernet:365]:   Subnet: 255.255.255.0
[17:47:11][C][ethernet:366]:   Gateway: 192.168.1.1
[17:47:11][C][ethernet:375]:   DNS1: 192.168.1.116
[17:47:11][C][ethernet:376]:   DNS2: 0.0.0.0
[17:47:11][C][ethernet:397]:   MAC Address: 0C:8B:95:CF:B2:XX
[17:47:11][C][ethernet:402]:   Is Full Duplex: YES
[17:47:11][C][ethernet:407]:   Link Speed: 100
[17:47:11][C][ethernet:224]:   Power Pin: 5
[17:47:11][C][ethernet:226]:   MDC Pin: 23
[17:47:11][C][ethernet:227]:   MDIO Pin: 18
[17:47:11][C][ethernet:228]:   Type: IP101
[17:47:11][C][ethernet:229]:   PHY addr: 1
[17:47:11][C][mdns:112]: mDNS:
[17:47:11][C][mdns:113]:   Hostname: gl-s10-bt-proxy-cfb2bc
[17:47:11][C][ota:093]: Over-The-Air Updates:
[17:47:11][C][ota:094]:   Address: gl-s10-bt-proxy-cfb2bc.local:3232
[17:47:11][C][api:138]: API Server:
[17:47:11][C][api:139]:   Address: gl-s10-bt-proxy-cfb2bc.local:6053
[17:47:11][C][api:141]:   Using noise encryption: YES
[17:52:04][I][ota:113]: Boot seems successful, resetting boot loop counter.
[17:52:04][D][esp32.preferences:114]: Saving 1 preferences to flash...
[17:52:04][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[17:52:05][D][esp32_ble_tracker:246]: Starting scan...
[17:57:05][D][esp32_ble_tracker:246]: Starting scan...
[18:02:05][D][esp32_ble_tracker:246]: Starting scan...
[18:07:05][D][esp32_ble_tracker:246]: Starting scan...
[18:12:05][D][esp32_ble_tracker:246]: Starting scan...
[18:17:05][D][esp32_ble_tracker:246]: Starting scan...
[18:22:05][D][esp32_ble_tracker:246]: Starting scan...
```

bdraco, 2023-08-22:
It should be scanning for Bluetooth devices continuously (each scan runs back to back). This is expected.

JohnyBeGood, 2023-08-22:
> It should be scanning for Bluetooth devices continuously (each scan runs back to back). This is expected.

Ok, after I hit stop in that window it shows as offline? YAML also never changes afterwards, does it need to be like this?
Also, HA never finds any BLE devices.

[image]

[image]

JohnyBeGood, 2023-09-02:
> > HA never finds any BLE devices.
> 
> Which integration are you using in HA that provides support for the BLE device you want to integrate?

[image]
[image]

Xiaomi TH Sensor LYWSD03MMC is what I'm trying to integrate

[image]

After scanning GL-S10 constantly shows offline:

[image]

bdraco, 2023-09-02:
What do you see in the diagnostics for the config entry?

JohnyBeGood, 2023-09-02:
> What do you see in the diagnostics for the config entry?

Not sure where is diagnostics but when I try to install it this is what I get at end of the log:

`INFO ESPHome 2023.8.2
INFO Reading configuration /config/esphome/gl-s10-bt-proxy-cfb2bc.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing gl-s10-bt-proxy-cfb2bc (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/src/main.o
Linking /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.elf
RAM:   [==        ]  15.1% (used 49492 bytes from 327680 bytes)
Flash: [======    ]  60.5% (used 1110753 bytes from 1835008 bytes)
Building /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.bin"], ["/data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.elf"])
Wrote 0x120950 bytes to file /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 19.05 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of gl-s10-bt-proxy-cfb2bc.local
INFO  -> 192.168.1.245
INFO Uploading /data/gl-s10-bt-proxy-cfb2bc/.pioenvs/gl-s10-bt-proxy-cfb2bc/firmware.bin (1116496 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from gl-s10-bt-proxy-cfb2bc.local using esphome API
INFO Successfully connected to gl-s10-bt-proxy-cfb2bc.local
[20:18:23][I][app:102]: ESPHome version 2023.8.2 compiled on Sep  1 2023, 20:17:23
[20:18:23][I][app:104]: Project esphome.bluetooth-proxy version 1.0
[20:18:23][C][status_led:019]: Status LED:
[20:18:23][C][status_led:020]:   Pin: GPIO32
[20:18:23][C][logger:301]: Logger:
[20:18:23][C][logger:302]:   Level: DEBUG
[20:18:23][C][logger:303]:   Log Baud Rate: 115200
[20:18:23][C][logger:305]:   Hardware UART: UART0
[20:18:23][C][i2c.idf:061]: I2C Bus:
[20:18:23][C][i2c.idf:062]:   SDA Pin: GPIO15
[20:18:23][C][i2c.idf:063]:   SCL Pin: GPIO13
[20:18:23][C][i2c.idf:064]:   Frequency: 50000 Hz
[20:18:23][C][i2c.idf:067]:   Recovery: bus successfully recovered
[20:18:23][I][i2c.idf:077]: Results from i2c bus scan:
[20:18:23][I][i2c.idf:079]: Found no i2c devices!
[20:18:23][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Reset Button'
[20:18:23][C][gpio.binary_sensor:016]:   Pin: GPIO33
[20:18:23][C][gpio.output:010]: GPIO Binary Output:
[20:18:24][C][gpio.output:011]:   Pin: GPIO14
[20:18:24][C][gpio.output:012]:   Inverted: YES
[20:18:24][C][gpio.output:010]: GPIO Binary Output:
[... 38 lines cut ...]
[20:22:54][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:22:54][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[20:22:55][D][esp32_ble_tracker:246]: Starting scan...
[20:27:55][D][esp32_ble_tracker:246]: Starting scan...
`

The only option is to stop:

[image]

Here's my gl-s10-bt-proxy-cfb2bc.yaml

`
substitutions:
  name: gl-s10-bt-proxy-cfb2bc
  friendly_name: Bluetooth Proxy cfb2bc
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/gl-s10.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: VoitS/ICZRPC2YnNnXm/3k6CYmoKfS4SfPhWeVxxxx`

bdraco, 2023-09-02:
https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics

JohnyBeGood, 2023-09-02:
> https://www.home-assistant.io/docs/configuration/troubleshooting/#download-diagnostics

Ah, sorry. The diagnostics file for "Bluetooth Proxy cfb2bc" is 1,696 lines long and I do recognize lots of Bluetooth devices in my home network, including BLE "Xiaomi TH Sensor LYWSD03MMC" that has label `"name": "ATC_66552B",`

jesserockz, 2023-10-05:
Hello,

We have archived this repo as the files have moved to [esphome/firmware](https://github.com/esphome/firmware)

If you install the latest ESPHome version and your issue still exists, please open an issue there.

Cheers,
Jesse

closed 2023-10-05 by jesserockz