esphome/bluetooth-proxies #83: M5 Atom Lite as Bluetooth Proxy

AseKarlsson, 2023-05-26:
Hi,

Having issues to use M5 Atom Lite as Proxies.

Original issue/case here:
https://github.com/esphome/esphome/discussions/4827
But get no help, so try to post here as well.

Everything programs OK but I cannot see any proxying any traffic and reach has not increased.

Bought these units, https://shop.m5stack.com/products/atom-lite-esp32-development-kit?ref=NabuCasa
Programmed them using https://esphome.github.io/bluetooth-proxies/

[image]

This created the following code:
````
substitutions:
  name: atom-lite-2-3
  friendly_name: Atom Lite 2-3

packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: REMOVED

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
````
This gives the following in the log:
````
INFO Reading configuration /config/esphome/atom-bluetooth-proxy-84e624.yaml...
INFO Starting log output from atom-lite-2-3.local using esphome API
INFO Successfully connected to atom-lite-2-3.local
[18:11:16][I][app:102]: ESPHome version 2023.5.2 compiled on May 22 2023, 17:38:42
[18:11:16][I][app:104]: Project esphome.bluetooth-proxy version 1.0
[18:11:16][C][wifi:505]: WiFi:
[18:11:16][C][wifi:363]:   Local MAC: [redacted]
[18:11:16][C][wifi:364]:   SSID: [redacted]
[18:11:16][C][wifi:365]:   IP Address: [redacted]
[18:11:16][C][wifi:367]:   BSSID: [redacted]
[18:11:16][C][wifi:368]:   Hostname: 'atom-lite-2-3'
[18:11:16][C][wifi:370]:   Signal strength: -52 dB ▂▄▆█
[18:11:16][C][wifi:374]:   Channel: 6
[18:11:16][C][wifi:375]:   Subnet: [redacted]
[18:11:16][C][wifi:376]:   Gateway: [redacted]
[18:11:16][C][wifi:377]:   DNS1: [redacted]
[18:11:16][C][wifi:378]:   DNS2: [redacted]
[18:11:16][C][logger:301]: Logger:
[18:11:16][C][logger:302]:   Level: DEBUG
[18:11:16][C][logger:303]:   Log Baud Rate: 115200
[18:11:16][C][bluetooth_proxy:065]: Bluetooth Proxy:
[18:11:16][C][bluetooth_proxy:066]:   Active: YES
[18:11:16][C][safe_mode.button:022]: Safe Mode Button 'Safe Mode Boot'
[18:11:16][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[18:11:16][C][esp32_ble:238]: ESP32 BLE:
[18:11:16][C][esp32_ble:240]:   MAC address: [redacted]
[18:11:16][C][esp32_ble:241]:   IO Capability: none
[18:11:16][C][esp32_ble_tracker:591]: BLE Tracker:
[18:11:16][C][esp32_ble_tracker:592]:   Scan Duration: 300 s
[18:11:16][C][esp32_ble_tracker:593]:   Scan Interval: 320.0 ms
[18:11:16][C][esp32_ble_tracker:594]:   Scan Window: 30.0 ms
[18:11:16][C][esp32_ble_tracker:595]:   Scan Type: ACTIVE
[18:11:16][C][esp32_ble_tracker:596]:   Continuous Scanning: True
[18:11:16][C][mdns:108]: mDNS:
[18:11:16][C][mdns:109]:   Hostname: atom-lite-2-3
[18:11:16][C][ota:093]: Over-The-Air Updates:
[18:11:16][C][ota:094]:   Address: atom-lite-2-3.local:3232
[18:11:16][C][api:138]: API Server:
[18:11:16][C][api:139]:   Address: atom-lite-2-3.local:6053
[18:11:16][C][api:141]:   Using noise encryption: YES
[18:11:16][C][improv_serial:032]: Improv Serial:
[18:13:52][D][esp32_ble_tracker:246]: Starting scan...
````
If I move the units "Hi-Link HD2410 BLE" closer to my Yellow with Bluetooth support they are detected just fine.
Have I missed something obvious here?
Any suggestions to go forward?

Thanks,
AseKarlsson

janprive, 2023-05-29:
substitutions:
  name: atom-bluetooth-proxy-bfb4c8
  friendly_name: Bluetooth Proxy bfb4c8
  board: m5stack-atom

esp32:
  board: ${board}
  framework:
    type: esp-idf

packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

bluetooth_proxy:
  active: true
  cache_services: true

api:
  encryption:
    key: ## removed

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

This is my config. However it gives me an error while compiling. Still it works at the moment....

janprive, 2023-05-29:
the log detects my toothbrush:

[11:49:10][D][esp-idf:000]: W (318403011) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[11:49:10][D][esp-idf:000]: W (318403018) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[11:49:10][D][esp-idf:000]: W (318403022) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[11:49:10][D][esp-idf:000]: W (318403025) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[11:49:10][D][esp-idf:000]: W (318403051) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

AseKarlsson, 2023-05-29:
Thanks, will test!

janprive, 2023-05-29:
Does your bt device work ? Then remove the usb bt stick. Pull it out. When the proxy is ok bt should work!

AseKarlsson, 2023-05-29:
I have a HA Yellow with BT support, If I bring the sensor closer to the Yellow it will be detected, I need these Atoms as proxies to get better coverage.

janprive, 2023-05-29:
You did add the bt proxy as an integration in home-assistant?

AseKarlsson, 2023-05-29:
Just the ESPHome integration, is there something else to add?

janprive, 2023-05-29:
[image]

AseKarlsson, 2023-05-29:
Yeah, they are detected and imported, just don have the "Bluetooth Proxy" in the name:
[image]

AseKarlsson, 2023-05-29:
The units show 1 device and 2 entities:
[image]

janprive, 2023-05-29:
Safe mode and firmware should it be

AseKarlsson, 2023-05-29:
Thats right!

janprive, 2023-05-29:
I’m not sure about your hi-link devices… could it be their not compatible with the atoms?

AseKarlsson, 2023-05-29:
Dont realy know, they are called BLE so thought that. Could they have any setting that could affect proxying?

janprive, 2023-05-29:
Bt proxy has no settings as far as I know…

AseKarlsson, 2023-05-29:
If I would go with my mobile (with Bluetootch active) near the Atom, should I be able to see something in its logs?

janprive, 2023-05-29:
Don’t know…. Try it! I’ll try too! Very interesting!

janprive, 2023-05-29:
mobile does nothing. when pairing a bt 'door sensor' it works and is in the logs.

AseKarlsson, 2023-05-29:
Feels like I have to look more into the Hi-Link LD2410 LBE Sensors

janprive, 2023-05-29:
That's imho the only thing left...

AseKarlsson, 2023-05-29:
@janprive, thanks so much for beeing helpful!

jpearn, 2023-06-04:
Just wanted to drop by and add my thanks as I had issues too getting my M5 Atom Lites working. My config. ended up being :

substitutions:
  name: atom-bluetooth-proxy-000000
  friendly_name: Bluetooth Proxy 1
  board: m5stack-atom
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
bluetooth_proxy:
  active: true
  cache_services: true

Now connects / works fine !

AseKarlsson, 2023-06-04:
For me everything was OK all the time it seemed after all investigations. The problem was that i had 2 meters between BT Proxy and the LD2410. If I got it down to 1 meter everything worked fine. Not the BT coverage i hoped for to use in the house :(

bdraco, 2023-08-06:
The ld2410 are notoriously bad for range. There is a newer model now that fixes that but it's not yet supported

AseKarlsson, 2023-08-06:
Thanks for this info Nick,
Info on new modelnumber to look for?

bdraco, 2023-08-06:
HLK-LD2450

matthew73210, 2023-08-16:
Hey all, is this working flawlessly for anyone ?
Throught i commented but it seems that it didn't save, so here goes again.
All I'm getting is the following in the logs 

```
[08:39:49][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[08:39:49][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[08:39:49][V][bluetooth_proxy:058]: Proxying 2 packets
[08:39:49][V][esp32_ble:178]: (BLE) gap_event_handler - 3
```

And very verbose is giving me

```
\xffL
[00:13:08][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[00:13:08][V][bluetooth_proxy:058]: Proxying 1 packets
[00:13:08][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 258140539496429
  rssi: -89
  address_type: 1
  data: '\xffL
```

258140539496429 is a device i can find when doing a scan with my phone. I have the config as mentioned above.
From what i can tell it's not parsing the address properly and thus not appearing in HA.
Any ideas? Maybe I should open an issue on esphome?

matthew73210, 2023-08-27:
> Hey all, is this working flawlessly for anyone ? Throught i commented but it seems that it didn't save, so here goes again. All I'm getting is the following in the logs
>
> ``` [...]

Error on my part, it has to be a device that can be tracked in homeassistant. Not all devices are. I has hoping for HA to pickup all ble devices and track them but this isn't the case.

xraive, 2023-09-13:
I don't know if I should open another issue but I noticed that I have a similar issue when it comes to the log output of the bluetooth proxy.  All the hardware I'm using is from the recommeneded list in ESPHome and Home Asssistant.  

My current hardware
bluetooth proxy (ESPHome) - M5Stack Atom Lite
Bluetooth Adapter - ORICO BTA-403 (CSR8510A10)
iBeacon Tracker - [NRF52810 iBeacon]

Should the logs look like this?

```
[11:14:03][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[11:14:03][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[11:14:03][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[11:14:03][V][bluetooth_proxy:058]: Proxying 3 packets
[11:14:03][VV][api.service:345]: send_bluetooth_le_raw_advertisements_response: BluetoothLERawAdvertisementsResponse {
  advertisements: BluetoothLERawAdvertisement {
  address: 136770387481119
  rssi: -52
  address_type: 0
  data: '\033\xffu
```

jesserockz, 2023-10-05:
Hello,

We have archived this repo as the files have moved to [esphome/firmware](https://github.com/esphome/firmware)

If you install the latest ESPHome version and your issue still exists, please open an issue there.

Cheers,
Jesse

closed 2023-10-05 by jesserockz