esphome/bluetooth-proxies #96: Instalattion fails on M5Stack Atom Lite

Darkangeel-hd, 2024-08-05:
I tried using the web a few keeps ago and it worked like a charm, may have something to do with the recent changes in the project.

The error the webinstaller gives after selection of the serial port is "_Failed to download manifest_"
[image]

Behind the scenes, is trying to request this file `https://firmware.esphome.io/bluetooth-proxy/m5stack-atom-lite/manifest.json`
but it ends up in a 404.

If I try to use the generic ESP32, it then request 
`https://firmware.esphome.io/bluetooth-proxy/esp32-generic/manifest.json` which completes successfully with a 200 status and the installation can proceed.

Is there anything I can do to help / diagnose / fix this issue?

hamsando, 2024-08-06:
Same issue with the bluetooth-proxy for olimex esp32 poe iso. I raised https://github.com/esphome/firmware/issues/255.

jesserockz, 2024-08-06:
Fixed by #97 and releasing now. Sorry. I thought I had tested this, but clearly I didnt.

closed 2024-08-06 by jesserockz

jesserockz, 2024-08-06:
This is now fixed on the website

Darkangeel-hd, 2024-08-07:
> Fixed by #97 and releasing now. Sorry. I thought I had tested this, but clearly I didnt.

I can confirm it's fixed and working again, thanks!
Mistakes happens, don't beat you to it