esphome/bluetooth-proxies #107: Ready Made Bluetooth Proxy vs YAML

royf007, 2025-02-10:
When I install the Bluetooth Proxy from the ESP Home project website: https://esphome.io/projects/ , it works fine.  When I try to use the YAML from Github, the esp32_ble_tracker does not start.

I would like to be able to obtain the YAML that corresponds to the ready made project.  It seems to be located at: https://github.com/esphome/bluetooth-proxies/blob/main/esp32-generic/esp32-generic.yaml

Flashing my ESP32 from this YAML doesn't work, but the compiled binary does.  I noticed the YAML specifies a minimum version of ESP Home of 2024.6 .  Is the binary compiled against this version?

My Home Assistant is now at version 2024.11 .  Is it possible that something broke in later versions of ESP Home?

If someone can please provide the correct YAML that matches the ready-made version on: https://esphome.io/projects/ , I would appreciate it.

jesserockz, 2025-02-10:
You are correct and that is the YAML source of the firmware located on the website. The ESPHome version used to compile can be found here.

https://github.com/esphome/bluetooth-proxies/blob/3b4df52fdd21ee20b3f489d6d06220bc4d5380cd/.github/workflows/build.yml#L25

but...

The firmware on the website is deployed directly from this repository using GitHub actions and is triggered by the [releases](https://github.com/esphome/bluetooth-proxies/releases)
As you can see there has not been a release for a while in this repository and I will be making a new one after ESPHome 2025.2.0 is released next week as it comes with esp-idf upgrades.

I am personally using my bluetooth proxies with ESPHome 2025.2.0-dev and they are working as expected. 

Please post logs showing the actual error you are seeing as "the esp32_ble_tracker does not start" does not describe the issue enough.

royf007, 2025-02-11:
Thank you for the quick response!  So, I decided to start fresh using the latest YAML source.  This time everything went like butter.

It wasn't so much that I was receiving any errors specific before, but there were no messages from the BLE_Tracker in the logs saying it was active, and it wasn't being discovered in HA as a bluetooth proxy.  All good now!

closed 2025-02-11 by royf007