esphome/bluetooth-proxies #108: STP Loop Error

marksims432, 2025-02-10:
Hi -

I have recently installed the ESP Home Bluetooth Proxy on an ESP32-POE-ISO-EA-IND and when I connect the device to my Unifi POE switch I am receiving an STP loop error.

Has anyone encountered this error and if they have could the method to resolve be shared?

I have previously installed and connected a similar ESP32-POE-ISO-EA-IND to my Unifi POE switch without issue and connected successfully to Home Assistant.

jesserockz, 2025-02-10:
This sounds like an issue that is not related to the firmware itself and is probably something wrong with the hardware.

closed 2025-02-10 by jesserockz

marksims432, 2025-02-16:
I was installing the ESP Home Bluetooth Proxy from my Macbook - when I reinstalled from my PC - it worked ok -