esphome/bluetooth-proxies #110: HA aaking to update to 2025.2.2

steps56, 2025-03-06:
Hello,
I got an warning to update to 2025.2.2, but can't get/find 2.2 still.
Waring says, to update m5stack-atom-ligte with ESPHome 2025.2.2 or higher.
How to do ?
Thanks for any help.

sarruman, 2025-03-07:
> Hello, I got an warning to update to 2025.2.2, but can't get/find 2.2 still. Waring says, to update m5stack-atom-ligte with ESPHome 2025.2.2 or higher. How to do ? Thanks for any help.

Same for me... Just searching if someone knows a way out of this...

steps56, 2025-03-07:
I found this:
https://community.home-assistant.io/t/update-esp32-bluetooth-proxy/859462/8

Now I'm updating using my laptop and plugged in the atom-lite by usb..
Cross your fingers for my success may be.
I will inform about here.

sarruman, 2025-03-07:
@steps56 Took my chances and reflashed the m5stack device using this page https://esphome.io/projects/?type=bluetooth.
So far, everything is working fine again.
If you haven't changed anything after initially flashing yours, I believe this is the easiest method!
Best of luck...

steps56, 2025-03-07:
> [@steps56](https://github.com/steps56) Took my chances and reflashed the m5stack device using this page https://esphome.io/projects/?type=bluetooth. So far, everything is working fine again. If you [...]

Yes, but this is only way I got it working again.
All those tryings about update doesn't work.
So still the hint at HA to update the atom-lite via ESP Home, but there no way, always 2025.2.1 and showing up to date, no update possible.

Going in circles and locked to the old firmware ... sadly.

Did get it updated ?
... and if, how?

steps56, 2025-03-07:
> not updated, just reflashed and got the latest firmware...

Reflash = 2025.2.1...is this latest ???
If so, why still this warning about to update to latest ???

steps56, 2025-03-07:
> > not updated, just reflashed and got the latest firmware...
> 
> Reflash = 2025.2.1...is this latest ??? If so, why still this warning about to update to latest ???

How to reflash ?
Bay the website?
Offering always lnly 2025.2.1 ...

sarruman, 2025-03-07:
🤷

steps56, 2025-03-07:
> 🤷

How did you "flash" ???
By ESPHomeBuilder ?
By Website ?
By HA/Integration ?

Just Website "flahing" gives a working Setup, all the rest just not working or even not possible via HA.

So, why 🤷🏼‍♂️ instead of explaining ??

nadir-ishiguro, 2025-03-07:
If you have a look at the only [other issue in this tracker,](https://github.com/esphome/bluetooth-proxies/issues/46)  you'll find this comment:

> This repository is only for hosting the [bluetooth-proxies](https://esphome.github.io/bluetooth-proxies/) web-installer site. Please only post here issues related to the installer functionality, or the website itself.
> 
> For bug reports related to bluetooth_proxy component and ESPHome issues, please go to [https://github.com/esphome/issues](https://github.com/esphome/issues/issues?q=sort%3Aupdated-desc+is%3Aissue+is%3Aopen)
> 
> For feature requests related to bluetooth_proxy component, go to [https://github.com/esphome/feature-requests](https://github.com/esphome/feature-requests/issues?q=sort%3Aupdated-desc+is%3Aissue+is%3Aopen)
> 
> For fast answers to direct questions, head on to our [Discord](https://discord.gg/KhAMKrd) support channel.
> 
> Thank you!

You're in the wrong place if you want help.

steps56, 2025-03-07:
Crossed about answer.
This Link gives no solution, as ESPHome Builder takes my half night to always end up by updated, but death proxy.
No way to get this way guiding to any working end.

But thanks again for adding/edditing your answer.
I will wait patiently till may be an update will pop up at HA in future.
That's a big diaadvantage of HA included stuff.
Any Integration and by console even HA maybe downgraded (or upgraded), but I'm to uninformed about  that bluetooth proxy thing.
(since Trump, I hate to write/talk in English, as anybody good minded should / may see the same)

But thanks again for info and communication, now I know its not my fault or dumbness 😎👍.

cweakland, 2025-03-13:
I had the same issue, so I decided to take control of my ESP devices and manually configure them. It was pretty easy to solve by installing the Home Assistant add-on "ESPHome Builder", be sure to tic the box "add to the sidebar". Open ESPHome Builder from the sidebar in Chrome browser.  Hit the secrets button and enter wifi information like so:

```
# Your Wi-Fi SSID and password
wifi_ssid: "your-ssid"
wifi_password: "your-password"
```

Connect your m5stack-atom device via USBC. Add a device and give it a name. It will take a few minutes to install. This looks like it puts a somewhat empty configuration on the device. 

For a bluetooth proxy I used the config from the ready made projects source:

https://raw.githubusercontent.com/esphome/bluetooth-proxies/refs/heads/main/m5stack/m5stack-atom-lite.yaml

Mine looked like this:

```
esphome:
  name: atom-bluetooth-proxy-b7a550
  friendly_name: Bluetooth Proxy - Play Room
  min_version: 2025.2.1

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
  encryption:
    key: "generate-an-api-key-and-put-it-here"
logger:

ota:
  - platform: esphome
    id: ota_esphome

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

captive_portal:
```

PR #111 (2025-03-14): Bump ESPHome to 2025.2.2
Closes #110
closed 2025-03-14 by frenck