esphome/bluetooth-proxies #122: gl-s10 configuration using depreciated "clk_mode"

bikeymouse, 2025-07-20:
Hope this is the proper location to post this issue as [the combined issue tracker is closed according to this message](https://github.com/esphome/issues/issues/7236) but I couldn't find a specific other (component) repo to post the issue?

Anyway:

I get the following warning when using https://github.com/esphome/bluetooth-proxies/blob/main/gl-inet/gl-s10.yaml

`WARNING [ethernet] The 'clk_mode' option is deprecated and will be removed in ESPHome 2026.1. Please update your configuration to use 'clk' instead.`

The Ethernet component configuration:

```
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk:
    pin: GPIO0
    mode: CLK_EXT_IN
  phy_addr: 1
  power_pin: GPIO5
```
Should be modified to:

```
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk:
    pin: GPIO0
    mode: CLK_EXT_IN
  phy_addr: 1
  power_pin: GPIO5
```

jesserockz, 2025-08-08:
Yes your proposed change is correct. Please open a pull request with those changes so everyone can benefit =)

bikeymouse, 2025-08-08:
Would like to contribute on top of telling what I did, but I'm sorry, I have never done such a thing so not sure how to do that.

Tried to follow instructions from Github help pages, but it seems I have to create a branch first which seems I'm not authorized to.

PR #123 (2025-08-20): Update deprecated clk_mode to clk configuration
The `clk_mode` option in ethernet configuration has been deprecated in favor of the new `clk` mapping format. This PR updates all affected configuration files to use the current ESPHome syntax and ensures build compatibility.

## Changes
- Updated gl-inet/gl-s10.yaml: GPIO0_IN → CLK_EXT_IN
- Updated olimex/olimex-esp32-poe-iso.yaml: GPIO17_OUT → CLK_OUT  
- Updated lilygo/lilygo-t-eth-poe.yaml: GPIO17_OUT → CLK_OUT
- Updated wt32/wt32-eth01.yaml: GPIO0_IN → CLK_EXT_IN
- Bumped min_version from 2025.5.0 to 2025.7.0 for all ethernet configs
- Updat [... cut]